PX4/PX4-Autopilot #3394: MC: Takeoff always results in slight yaw in SITL

LorenzMeier, 2015-12-28:
@thomasgubler Is this related to your yaw fix? I checked and its not a declination issue - the yaw setpoint is non-zero on takeoff waypoints, which I don't understand yet fully.

thomasgubler, 2015-12-28:
Unless you are flying with MIS_YAWMODE != 0 this is not related.

LorenzMeier, 2015-12-28:
We are by default facing the next waypoint.

thomasgubler, 2015-12-28:
ah ok I see, MIS_YAWMODE == 1 per default. Then this could be related. Although I did not touch the takeoff behaviour (which is "do nothing" https://github.com/PX4/Firmware/blob/master/src/modules/navigator/mission.cpp#L531 )

thomasgubler, 2015-12-28:
For me it's non-zero because of https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L1634

LorenzMeier, 2015-12-28:
Fixed.

closed 2015-12-28 by LorenzMeier