PX4/PX4-Autopilot #3400: VTOL ALTCTL transition: no throttle value

melisab, 2015-12-29:
After completing transition in standard VTOL in ALTCTL mode the plane started dropping and user cannot control the fw throttle for awhile.

http://logs.uaventure.com/view/ZEkEPTKrDpNocfQrvUp7h6
http://logs.uaventure.com/view/gKmtA9YSR7jc9quZ6RSSoX

LorenzMeier, 2016-01-01:
Thanks for sharing these logs! We are looking into these.

LorenzMeier, 2016-01-01:
@tumbili Initial analysis shows that the system indeed commanded zero throttle. I'm not sure yet why.

melisab, 2016-01-02:
Just want to add another thing from my observation. In ALTCTL the plane also cannot maintain its altitude during transition; i.e. mc throttle shot up during transition.

closed 2016-02-14 by tumbili

melisab, 2019-03-13:
Hi! I started flying the same VTOL plane again. Encounter the same issue in  POS CTL. I was wondering if the issue has been found out with recent firmware? #11628