PX4/PX4-Autopilot #3415: Windows USB driver: Add VID / PID for Pixracer to Windows drivers

LorenzMeier, 2016-01-01:
The MSI has to be added here: https://pixhawk.org/firmware/downloads

LorenzMeier, 2016-01-01:
String: "PX4 BL FMU v4.x"
USB PID: 0x26AC
USB VID: 0x0012

@meee1 It would be fantastic if you could help us out with another MSI installer which contains this. It would be also great to have this in the MSI for all the other software you're redistributing!

LorenzMeier, 2016-01-01:
@kd0aij This is for Windows.

LorenzMeier, 2016-01-01:
@DonLakeFlyer Windows users will need the new drivers for Pixracer to work via USB.

meee1, 2016-01-01:
currently the msi targets these devices
DriverInstall, USB\VID_26AC&PID_0001, USB\VID_26AC&PID_0002, USB\VID_26AC&PID_0003, USB\VID_26AC&PID_0010, USB\VID_26AC&PID_0011, USB\VID_26AC&PID_0012, USB\VID_26AC&PID_0013, USB\VID_26AC&PID_0014
of which the new one is already in the list, only the description is incorrect, but should work as it stands.

do we want a new msi that corrects the name?

LorenzMeier, 2016-01-01:
hmm, let me re-confirm with our Windows users what issue they were seeing. Thanks!

LorenzMeier, 2016-01-02:
@nickarsov could you provide a screenshot of the error you were getting? I'm assuming you have the drivers from here installed? https://pixhawk.org/firmware/downloads

LorenzMeier, 2016-01-02:
Ok, I think I found the error: The application had the wrong product ID. That of course doesn't go well. Its fixed in bbe69fb.

closed 2016-01-02 by LorenzMeier