PX4/PX4-Autopilot #3421: mavros can't communicate to jmavsim(DrTon version) on ubuntu 14.04 LTS

lishidetiankong, 2016-01-03:
Hello developers of pixhawk:
      Now, I try to use hil mode to test px4 hign level logic, so I want to use mavros and jmavsim together with pixhawk hardware to implement the test process, and monitor with QGC2.8.0
      First I use jmavsim and pixhawk without mavros, this process work successfully, I can use RC control pixhawk and fly in the jmavsim. 
     
 Next, I test connect pixhawk to mavros without jmavsim(I modify px4.launch as below two line shown
	<arg name="fcu_url" default="/dev/ttyACM0:57600" />
	<arg name="gcs_url" default="udp://:14556@127.0.0.1:14550" />)
       And this process work successfully too, because of no jmavsim paticipate in, so I can't watch how px4 fly ,Only could watch some message using QGC
  
      But , I meet a key problem, that is mavros can't communicate to jmavsim(DrTon version)

lishidetiankong, 2016-01-03:
I can make sure the target ID of mavlink on the pixhawk is the same with px4.launch of mavros

LorenzMeier, 2016-01-03:
Please follow these instructions:
http://dev.px4.io/simulation-ros-interface.html

And make sure to use latest PX4 master.

closed 2016-01-03 by LorenzMeier