PX4/PX4-Autopilot #3425: gazebo problem package 'gazebo>=6.0' not found

won13y, 2016-01-03:
I did as the following page describes: 

http://dev.px4.io/simulation-gazebo.html

but I got the error

I seems to me that the version of installed gazebo is not gazebo>=6.0'

--------------------------------------------------------------------------------------------------------------------------

wkyoun@wkyoun-XPS-13-9343:~/catkin_ws/src/Firmware$ make posix_sitl_default gazebo

make -C /home/wkyoun/catkin_ws/src/Firmware/build_posix_sitl_default -j4 --no-print-directory gazebo
[  2%] Built target df_driver_framework
[  2%] Built target git_genmsg
[  3%] Built target git_gencpp
[  3%] Built target prebuild_targets
[  3%] Built target git_mavlink
[  4%] Built target git_matrix
[  4%] Built target git_uavcan
[  5%] [  5%] Built target generate_topic_listener
Built target git_driverframework
[  6%] [  6%] Built target xml_gen
Built target mixer_gen
[  6%] Built target git_jmavsim
[  7%] Built target msg_gen
[  7%] Built target platforms__common
[  7%] Built target git_gazebo
[  8%] [  8%] Built target drivers__boards__sitl
Built target drivers__pwm_out_sim
[ 12%] Built target drivers__device
[ 12%] Built target platforms__posix__drivers__adcsim
[ 15%] [ 22%] Built target platforms__posix__px4_layer
Built target platforms__posix__work_queue
[ 23%] Built target platforms__posix__drivers__tonealrmsim
[ 23%] Built target platforms__posix__drivers__gpssim
[ 24%] [ 25%] Built target platforms__posix__drivers__accelsim
Built target platforms__posix__drivers__airspeedsim
[ 25%] Built target platforms__posix__drivers__rgbledsim
[ 26%] Built target platforms__posix__drivers__barosim
[ 28%] [ 28%] Built target platforms__posix__drivers__ledsim
Built target systemcmds__mixer
[ 28%] Built target platforms__posix__drivers__gyrosim
[ 28%] Built target systemcmds__param
[ 29%] Built target systemcmds__ver
[ 29%] Built target systemcmds__esc_calib
[ 29%] Built target systemcmds__reboot
[ 30%] Built target systemcmds__topic_listener
[ 31%] Built target systemcmds__perf
[ 32%] Built target modules__param
[ 37%] [ 37%] Built target modules__uORB
Built target modules__sensors
[ 40%] Built target modules__systemlib__mixer
[ 41%] Built target modules__ekf_att_pos_estimator
[ 42%] Built target modules__simulator
[ 43%] [ 44%] Built target modules__attitude_estimator_ekf
Built target modules__attitude_estimator_q
[ 47%] Built target modules__vtol_att_control
[ 47%] Built target modules__mc_pos_control
[ 52%] [ 54%] Built target modules__mavlink
Built target modules__position_estimator_inav
[ 55%] Built target modules__mc_att_control
[ 57%] [ 63%] [ 65%] Built target modules__mc_att_control_multiplatform
Built target modules__navigator
Built target modules__mc_pos_control_multiplatform
[ 68%] [ 70%] [ 71%] Built target modules__land_detector
Built target modules__controllib
[ 72%] Built target lib__external_lgpl
Built target modules__fw_att_control
[ 74%] [ 74%] Built target modules__dataman
Built target lib__mathlib
[ 75%] Built target modules__sdlog2
[ 78%] Built target lib__ecl
[ 78%] Built target lib__mathlib__math__filter
[ 79%] Built target lib__conversion
[ 79%] [ 85%] Built target lib__geo
Built target modules__commander
[ 85%] Built target lib__terrain_estimation
[ 86%] Built target lib__geo_lookup
[ 87%] Built target lib__launchdetection
[ 88%] Built target lib__runway_takeoff
[ 96%] Built target modules__systemlib
[ 98%] Built target modules__fw_pos_control_l1
[100%] Built target mainapp
SITL ARGS
rc_script: posix-configs/SITL/init/rcS
debugger: none
program: gazebo
model: none
build_path: /home/wkyoun/Firmware/build_posix_sitl_default
empty model, setting iris as default
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   thread
-- checking for module 'gazebo>=6.0'
--   package 'gazebo>=6.0' not found
CMake Error at /usr/share/cmake-3.2/Modules/FindPkgConfig.cmake:344 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.2/Modules/FindPkgConfig.cmake:506 (_pkg_check_modules_internal)
  CMakeLists.txt:7 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
See also "/home/wkyoun/Firmware/Tools/sitl_gazebo/Build/CMakeFiles/CMakeOutput.log".
make[4]: *** [src/firmware/posix/CMakeFiles/gazebo] Error 1
make[3]: *** [src/firmware/posix/CMakeFiles/gazebo.dir/all] Error 2
make[2]: *** [src/firmware/posix/CMakeFiles/gazebo.dir/rule] Error 2
make[1]: *** [gazebo] Error 2
make: *** [posix_sitl_default] Error 2
wkyoun@wkyoun-XPS-13-9343:~/catkin_ws/src/Firmware$ 

---------------------------------------------------------------------------------------------------------------------------

So, I tried to install gazebo6 as following web:

http://gazebosim.org/tutorials?tut=install_ubuntu&

In Step-by-step Install

When I did following command:

I got the error like this

How can I fix this?

wkyoun@wkyoun-XPS-13-9343:~$ sudo apt-get install gazebo6

Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 gazebo6 : Depends: libgazebo6 (= 6.5.1-1~trusty) but it is not going to be installed
           Depends: libsdformat3 but it is not going to be installed
           Depends: gazebo6-common (= 6.5.1-1~trusty) but it is not going to be installed
           Recommends: gazebo6-plugin-base
E: Unable to correct problems, you have held broken packages.
wkyoun@wkyoun-XPS-13-9343:~$

andre-nguyen, 2016-01-03:
Try uninstalling your current gazebo version first.

closed 2016-01-06 by LorenzMeier

won13y, 2016-01-06:
How Can I check the current version of gazebo?
I tried to search the method to check the current version of gazebo, but nothing found

andre-nguyen, 2016-01-06:
~$ gazebo --version
    Gazebo multi-robot simulator, version 6.5.1
    Copyright (C) 2012-2015 Open Source Robotics Foundation.
    Released under the Apache 2 License.
    http://gazebosim.org

    Gazebo multi-robot simulator, version 6.5.1
    Copyright (C) 2012-2015 Open Source Robotics Foundation.
    Released under the Apache 2 License.
    http://gazebosim.org