PX4/PX4-Autopilot #3428: quad x8，I can't find it

DavidsWONG, 2016-01-04:
why i can't find quad x8？
how can i add it myself？
thanks，

DavidsWONG, 2016-01-04:
also I can not find Y6 frame！how can i do it myself。the Documentation of pixhawk.org show none tip for it.

LorenzMeier, 2016-01-04:
Just scroll down in QGroundControl. What you're looking for is the Octarotor X coaxial.

LorenzMeier, 2016-01-04:
Need to add all frames to docs.

DavidsWONG, 2016-01-05:
@LorenzMeier   
I found 19 frames of QGroundControl 2.8.0. But I can't found the frame named Octarotor X coaxial.

SimonWilks, 2016-01-05:
@DavidsWONG do you not see this one in your selection? The image of the plane that goes along with it is obviously wrong but that's the one you'll need.
[image]

LorenzMeier, 2016-02-21:
Fixed through QGC's auto-generated list now.

closed 2016-02-21 by LorenzMeier