PX4/PX4-Autopilot #3431: Write BST telemetry user setup guide

LorenzMeier, 2016-01-04:
@SirPerna @dogmaphobic It would be fantastic to cover this in the PX4 user guide. It auto-starts, so its mostly about wiring. http://px4.io/user-guide/ ? We want to finish this asap.

LorenzMeier, 2016-01-04:
We got the wiring here:
https://pixhawk.org/peripherals/telemetry/crossfire

We need to make up our mind how to document - either just describe what it is and where to buy and link to that page for wiring, or duplicate the page.

dogmaphobic, 2016-01-04:
I just need to wrap my head around the new web site. For instance, when I go to the page above and try to edit it (there is a small typo in it), instead of having content, it points to something else (```[docit id=di_px4_autopilot intro=746]```). When I go to *Doc It*, which is where I assume these documents are located, there is no clear indication of which page/document holds it.

I need to finish the work with logging first. I can then spend some time learning the new web site. I also have yet to get the Crossfire to work properly using the new *UART* link, which is very different from the wiring described above.

LorenzMeier, 2016-01-04:
There is a PX4 User guide category in Doc it. That's where all the documentation is.

dogmaphobic, 2016-01-04:
Yes, I figured there would be, I just can't find it :)

[image]

LorenzMeier, 2016-01-04:
Sorry, I meant documents. They are all there.

dogmaphobic, 2016-01-04:
I'm certainly missing something. I still can't find it:

[image]

LorenzMeier, 2016-01-04:
You have all pages there. Obviously there is no Crossfire page. What are you looking for?

dogmaphobic, 2016-01-04:
I was looking specifically for the page mentioned above (http://px4.io/user-guide/). It has a small typo:

_This guide is brand new and will be completed by the of January 2016. Right now it might be suitable to get started, but does not yet cover all of PX4._

Should be:

_This guide is brand new and will be completed by the **end** of January 2016. Right now it might be suitable to get started, but does not yet cover all of PX4._

LorenzMeier, 2016-01-04:
See the introductions menu.

dogmaphobic, 2016-01-04:
Ha! Found it. That's what I meant. If you go to a page, there seems to be no indication of where its content is located. If you go to http://px4.io/user-guide and *edit it*, all you see is a cryptic tag: ```[docit id=di_px4_autopilot intro=746]```. No direct link to where the content actually is.

Fixed the typo. As I said, let me get this logging thing done first. Then I can spend some time learning the new web site. The question of what is the proper way to connect the Crossfire remains though.

closed 2016-02-18 by LorenzMeier