PX4/PX4-Autopilot #3432: VTOL: Enforce transition switch

LorenzMeier, 2016-01-04:
AUX1 check

LorenzMeier, 2016-01-04:
@tumbili Check the switch here: https://github.com/PX4/Firmware/blob/master/src/modules/commander/PreflightCheck.cpp#L524

sanderux, 2016-01-11:
@LorenzMeier some people might not have a transition switch configured when tuning the quad. (setting a knob for PID tuning). Or have another reason for not assigning the switch.

sanderux, 2016-03-06:
@AndreasAntener considering my comment; yes or no on this?

AndreasAntener, 2016-03-10:
On a different matter: we should make this an actual switch (instead of an aux channel) and evaluate it also in the RC main state handling. At the moment the flight modes don't get re-evaluated when you flick the transition switch.

AndreasAntener, 2016-03-10:
And I don't think it's much of a deal breaker if you need this switch. People who do PID tuning with a knob sure have a 16chan radio

LorenzMeier, 2016-07-29:
Done.

closed 2016-07-29 by LorenzMeier