PX4/PX4-Autopilot #3435: Document offboard / straight pass through startup

LorenzMeier, 2016-01-04:
This will be popular for ROS users.

Sending:
set_actuator_control_target

Receiving:
rc_channels

LorenzMeier, 2016-01-04:
Being implemented here: https://github.com/PX4/Firmware/pull/3433

@julianoes This serves as interface of the Snapdragon Flight to hardware

dagar, 2017-04-11:
Still needed?

mhkabir, 2017-04-11:
We just need to document it. It has been working for ages.

julianoes, 2017-04-12:
I would vote to remove it since Snappy now supports PWM output and Spektrum input and it's kind of a clunky setup.

mhkabir, 2017-04-12:
The offboard actuator control interface, even if not used for Snapdragon is very popular with ROS people and others who implement their controllers offboard.

julianoes, 2017-04-12:
@mhkabir fair enough :)

closed 2018-02-22 by PX4BuildBot