PX4/PX4-Autopilot #3438: USB serial port connection only working if serial port connection is opened during a tiny time window while booting the pixhawk

molysgaard, 2016-01-04:
I am having lots of trouble getting the USB link working efficiently.
I have to open the serial connection at a (empirically measured) 1 sec window during the boot process.
Before this window the serial port is not available (not listed in /dev/) and after this opening the serial port works, but there is never any data sent or received.

My hunch is that this is some how related to the hardware in my laptop, because my colleagues do not have the same issue.

I am using a 3DR pixhawk.
I have a stock Thinkpad t440 with intel chipset and Intel USB3.0 ports.
I am running Ubuntu 15.04 (vivid)
The attached file contains the output of `lshw` giving the exact hardware contained in the laptop [lshw output](https://github.com/PX4/Firmware/files/77293/lshw.txt)

I have tried the following solutions:
* [Disable TLP USB-autosuspend](http://linrunner.de/en/tlp/docs/tlp-configuration.html#usb)
* Power the laptop externally
* Power the pixhawk externally and connect via usb after boot
* Tested all USB-ports on the laptop

estechnical, 2016-01-07:
Is it possible that modem manager is installed? This does some weird things to usb serial devices when they are first attached and if you missed the step "apt-get remove modemmanger" you will probably have this sort of issue connecting. I hope that helps...

molysgaard, 2016-01-07:
@estechnical I forgot to mention that I have removed modemmanager.

LorenzMeier, 2016-05-22:
This must be due to your local setup.

closed 2016-05-22 by LorenzMeier