PX4/PX4-Autopilot #3449: MC: reduce deadband and make thrust control smoother in position/altitude control

AndreasAntener, 2016-01-05:
Currently the deadband is too big and it reacts to aggressive on throttle changes (POSCTL/ALTCTL)

kd0aij, 2016-01-05:
I've been saying that for a while now. Also see https://github.com/PX4/Firmware/pull/3418

AndreasAntener, 2016-01-05:
Hey Mark, yep, saw your PR. But the 2-point curve is for manual? I won't touch anything there, just what you already did with the deadband plus trying to low-pass thrust input for the assisted modes.

kd0aij, 2016-01-05:
still experimenting with the manual throttle curve. But I do think crashing into the edge of the ALTCTL deadband is most of the problem. It seems as though the z PID loop has some built-in hysteresis too, but I haven't looked at the controller yet. My guess is that a properly designed and tuned controller will feel just fine with a linear mapping for throttle stick to vertical speed setpoint. (And no mucking with lowpass filters on the control input, which will add control latency)

kd0aij, 2016-01-05:
BTW, if I'm not confused, the 2-point curve in my PR applies to the ALTCTL throttle mapping, not manual

AndreasAntener, 2016-01-05:
I agree, as far as I tested already the deadband is most of the problem. Did your RP already resolve that for you?

Tuning matters of course. Another option would be to do acceleration control like we do in horizontal now. I think the control latency in pos/altctl for a low-pass over a short time could be disregarded.

kd0aij, 2016-01-05:
My goal is to make ALTCTL better than manual mode for FPV, hence the goal is zero latency and precision control of vertical speed.
Regarding resolution of the problem, I think the video in #3418 looks and sounds much better. Also note there is no transient on switching from manual to ALTCTL.

kd0aij, 2016-01-05:
I hope it's not necessary to define a new ALTCTL mode for FPV, but that will be necessary if non-fpv users want "sloppy" controls

AndreasAntener, 2016-01-05:
I think there are 3 levels: sloppy, ok for average Joe, and direct control for racing pilots. Sloppy we should never have. So my question would be, can you get altitude control (from the estimation to the output) as good as it would be with direct input control for people who really need it? I'm assuming your trying to figure this out at the moment :) (which would then answer your last question as well)

Anyway, I just looked at your throttle curve which made me assume it's for manual only, but the deadband change affects alt/posctl of course. I'll give it go.

kd0aij, 2016-01-05:
It seems reasonable to expect that the autopilot would have a significant advantage over the pilot (LOS or FPV) for reactive control of vertical speed/acceleration, so I'm optimistic that we can provide a real advantage to the FPV racing pilot with ALTCTL. But the requirements for FPV racing may not be compatible with those of other users who want less authority over the vehicle.

AndreasAntener, 2016-01-05:
I see what you mean and I agree that in terms of control the autopilot should be better than the pilot. I'm mostly concerned with altitude estimation as that it just won't be good enough. I'm happy to be proven wrong :) because perfect altitude estimation is something every user wants (no matter what he does).

As for racing requirements I'm not sure about the deadband in general. If the pilot changes throttle it should react instantly (no deadbadn). If he let's it sit near 50% it should hold altitude (small deadband). Is that what your 2-point curve does?

kd0aij, 2016-01-05:
That was the idea, just a quick & dirty form of expo. But it feels best to me with zero deadband and linear response across the entire stick range. However, that's flying beta Pixracer hardware. I'll test later today on fmuV2 with a proper static port for the baro. Testing is a pain since you have to recompile/reflash to change the ALTCTL throttle behavior.

tumbili, 2016-01-05:
@kd0aij @AndreasAntener How about something like this

XXX                              XXX
   XXX                        XXX
     XXX                   XXX
       XXX               XXX
         XXX          XXX
             XXXXXXX

Linear with a small dead-band. The only difference would be that the signal will not jump after the dead-band but increase in a linear fashion.

tumbili, 2016-01-05:
@AndreasAntener @kd0aij Sorry, the editor messed up my excellent drawing :-(

kd0aij, 2016-01-05:
Hi Roman. Not sure I understood your plot, but the 2-point curve is supposed to eliminate the slope discontinuities at +/-alt_ctl_dz. Below is a sketch of my intention.
Just flew #3418 on my bigger quad and there is definitely still some deadband in ALTCTL throttle input. That implies a bug or error in my code change, since I intended zero deadband. Will have to put some prints back in to double-check my arithmetic, then look for hysteresis elsewhere if that's not it.
[image]

kd0aij, 2016-01-11:
Looking closer at my Aeroquad log, it seems that the deadband/hysteresis is coming from the altitude control loop, not stick inputs.

tumbili, 2016-01-11:
@kd0aij Will send you a picture of another suggestion today

tumbili, 2016-01-11:
@kd0aij How about something like this?
[image]

kd0aij, 2016-01-11:
That's the way it was to start with

tumbili, 2016-01-13:
@kd0aij hahaha, good one :-) Any comments on the upper curve?

tumbili, 2016-01-13:
This would make sure that the user has a deadzone and it would also make sure that the desired velocity does not jump after the deadzone. Of course the slope of the curve is not exactly 45 degrees any more but this shouldn't be problematic.

kd0aij, 2016-01-13:
your sketch is the way I thought the curve looked to start with. Mine reduces the slope discontinuity at the break by adding the dy parameter

kd0aij, 2016-01-13:
note the (ctl - dz) terms in scale_control()

kd0aij, 2016-01-13:
the problem with a large deadband is the pilot doesn't know where the edge is. I typically wind up with the stick near one edge of the deadzone, then have to move it a long way to correct in the opposite direction. As a result the stick is moving quickly when it leaves the deadband and the first thing the pilot notices is the sudden increase in thrust. Same as having play in an automobile's steering wheel, but worse because there's no feedback from the stick to tell you when it's actually doing something.

tumbili, 2016-01-13:
@kd0aij But with your method the user will have to place the stick very precisely in order to have zero vertical movement of the quad. I agree that at the moment the pilot feels a sudden increase of thrust but in my sketch the desired velocity starts at zero again after the dead zone so the thrust increase should not jump.

tumbili, 2016-01-13:
@kd0aij You have to mentally rotate my picture by 90 degrees anti clockwise.

kd0aij, 2016-01-13:
doesn't this implement your curve though?
```
float
MulticopterPositionControl::scale_control(float ctl, float end, float dz)
{
	if (ctl > dz) {
		return (ctl - dz) / (end - dz);

	} else if (ctl < -dz) {
		return (ctl + dz) / (end - dz);

	} else {
		return 0.0f;
	}
}
```

tumbili, 2016-01-13:
@kd0aij Yes, this is exactly it.

kd0aij, 2016-01-13:
Git Blame (from inside QtCreator... nice)
```
20c9ce9d (Anton Babushkin 2013-12-26 23:03:19 +0400) float
20c9ce9d (Anton Babushkin 2013-12-26 23:03:19 +0400) MulticopterPositionControl::scale_control(float ctl, float end, float dz)
20c9ce9d (Anton Babushkin 2013-12-26 23:03:19 +0400) {
20c9ce9d (Anton Babushkin 2013-12-26 23:03:19 +0400) 	if (ctl > dz) {
20c9ce9d (Anton Babushkin 2013-12-26 23:03:19 +0400) 		return (ctl - dz) / (end - dz);
20c9ce9d (Anton Babushkin 2013-12-26 23:03:19 +0400) 
20c9ce9d (Anton Babushkin 2013-12-26 23:03:19 +0400) 	} else if (ctl < -dz) {
20c9ce9d (Anton Babushkin 2013-12-26 23:03:19 +0400) 		return (ctl + dz) / (end - dz);
20c9ce9d (Anton Babushkin 2013-12-26 23:03:19 +0400) 
20c9ce9d (Anton Babushkin 2013-12-26 23:03:19 +0400) 	} else {
20c9ce9d (Anton Babushkin 2013-12-26 23:03:19 +0400) 		return 0.0f;
20c9ce9d (Anton Babushkin 2013-12-26 23:03:19 +0400) 	}
20c9ce9d (Anton Babushkin 2013-12-26 23:03:19 +0400) }
```

tumbili, 2016-01-13:
@kd0aij So this is actually what's already implemented then. Strange, doesn't really feel like it.

kd0aij, 2016-01-13:
my tests with zero deadband indicated that the altitude controller has its own deadband or hysteresis. But I haven't done any code inspection yet.

tumbili, 2016-01-13:
@kd0aij Interesting, let me check

tumbili, 2016-01-13:
@kd0aij Did you have a zero vel_z I gain?

kd0aij, 2016-01-13:
MPC_Z_VEL_I	0.02

tumbili, 2016-01-13:
@kd0aij I don't see any reason why there should be a deadband. Not sure about the resolution of the throttle stick though.

kd0aij, 2016-01-13:
maybe this log and video will tell you something: https://github.com/PX4/Firmware/pull/3418#issuecomment-168426989

I had dy=dz for that throttle curve, so it should have been linear with no deadband

tumbili, 2016-01-13:
@kd0aij I actually know now where the hysteresis comes from, see here
https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L800

If you input is too small then the quad will still  have vertical position locked and just stay on the exact same height.

tumbili, 2016-01-13:
@kd0aij Hey, this might be causing the non smooth behaviour.

kd0aij, 2016-01-13:
looks likely to be the cause. Am I stupid, or is that code just hard to follow?

kd0aij, 2016-01-13:
and yes it looks like it's another deadband around the current altitude

tumbili, 2016-01-13:
@kd0aij I agree, it's hard to follow :-D

tumbili, 2016-01-13:
@kd0aij Going for some sleep now, will catch up with you on this tomorrow.

kd0aij, 2016-01-13:
thanks, I'll see if I can flight test with that zeroed out

kd0aij, 2016-01-14:
@tumbili That was it! I zeroed that parameter and the quad responded to the throttle stick smoothly. Here's a plot with no deadband on throttle or z velocity demand:
[image]

Note that thrust tracks the rapid throttle changes closely. And I no longer observed hesitation or surges when accelerating into forward flight, or when asking for changes in vertical speed.

tumbili, 2016-01-14:
@kd0aij Nice! Note that the same things applies for horizontal velocities.

AndreasAntener, 2016-01-14:
Is it still entering alt hold if you leave the stick centered? Or do we have to make hold_z_dz just very small?

tumbili, 2016-01-14:
@AndreasAntener If we have the deadband from before we could make hold_z_dz very very small. But I'm still thinking what the proper fix would be.

tumbili, 2016-01-14:
@AndreasAntener If we do the same thing for the horizontal movement then we might also get rid of the twitches in pos control. Or have those been resolved already?

AndreasAntener, 2016-01-14:
The twitches have a different source. They also happen between stick changes when your not at 0, so not deadband related.

tumbili, 2016-01-14:
@kd0aij What do you think, we can we now go for the old version of the deadband or do you feel like your solution even feels better?

kd0aij, 2016-01-14:
I need to try smaller values of dy in my PR https://github.com/PX4/Firmware/pull/3418. So far I've been using a straight line curve.
I think for FPV racing, zero deadband will be preferred, and for most other uses, some deadband will be convenient (or else a centering spring on throttle).

kd0aij, 2016-01-14:
just flew with 
```
    const float alt_ctl_dz = 0.05f;
    const float alt_ctl_dy = 0.0f;
```
It's extremely windy here today, but the surges were noticeable even with the relatively small dz.
I'll try setting dy to .02 and see how that feels. Too bad we don't have the equivalent of OpenTX global variables for this sort of thing.

kd0aij, 2016-01-14:
altctl_ctl_dy = .02 is a definite improvement over  the zero case.
Sensitivity of vertical speed control near center stick feels like it's on the order of the drift in the baro altitude estimate. And I can generally tell when I'm moving the stick away from center by the vehicle's response. 

My opinion is that adding one or two parameters for dy and dz would be well worth it.

LorenzMeier, 2016-01-14:
Sounds very conclusive - do you want to have a go at it?

kd0aij, 2016-01-15:
I'll add alt_ctl_dy,dz parameters to https://github.com/PX4/Firmware/pull/3418  with defaults to match the current deadband

tumbili, 2016-01-15:
@kd0aij Cool, looking forward to test this.

kd0aij, 2016-01-16:
@tumbili https://github.com/PX4/Firmware/pull/3418 has the new parameters

tumbili, 2016-02-19:
Implemented in #3418

closed 2016-02-19 by tumbili