PX4/PX4-Autopilot #3451: Accelerometer bias correction using vision in position_estimator_inav

thedinuka, 2016-01-06:
I think the way the mocap based accel bias correction was added causes vision based bias correction to have no effect, whether or not mocap is being used. The problem is here:
https://github.com/PX4/Firmware/blob/master/src/modules/position_estimator_inav/position_estimator_inav_main.cpp#L1030-L1032

Bias correction values are being reset before they were actually incorporated into the accel bias estimate further down in code. 
Please note that I haven't been able to verify this through experiments yet. But this seems a very straightforward issue.

LorenzMeier, 2016-01-06:
It would be fantastic if you could send a PR with a fix in case this turns out to be required. You certainly seem to have a point based on the high-level analysis.

mhkabir, 2017-04-02:
Outdated.

closed 2017-04-02 by mhkabir