PX4/PX4-Autopilot #3452: Acceleration setpoints in offboard mode

julianoes, 2016-01-06:
The support for acceleration setpoints in offboard mode needs to be revisited/tested.

tuofeichen, 2016-02-12:
Hi @julianoes,
Did you test out the acceleration setpoint in offboard mode? I'm running into some problems using the mavros setpoint_accel plugins (see https://github.com/PX4/Firmware/issues/3743). I'm also wondering if the mavros plugin's not working, can I directly write to a position_setpoint msg, which includes the acceleration setpoint? https://github.com/PX4/Firmware/blob/master/msg/position_setpoint.msg

Thank you

Best Regards,
Francis

julianoes, 2016-02-13:
As far as I know, currently only position and velocity setpoints are supported. Acceleration setpoints are to do.
@AndreasAntener correct me if I'm wrong.

tuofeichen, 2016-02-13:
Hi @julianoes ,
Thanks for clarifying this, just out of curiosity, isn't setpoint_accel the easier one to implement, given that the IMU raw data is sufficient? The velocity and position setpoint would require some sort of additional information to correct the drift of integrating acceleration in the IMU?

Thanks!

Best Regards,
Francis

julianoes, 2016-07-25:
This has been implemented in #4010.

closed 2016-07-25 by julianoes