PX4/PX4-Autopilot #3453: Gdb segment fault with arm-gcc 4.9

blandry, 2016-01-06:
After compiling using gcc 4.9.3 (OSX) gdb segment faults when trying to read symbols from the elf:

    (gdb) file .../build_px4fmu-v2_default/src/firmware/nuttx/firmware_nuttx
    Reading symbols from .../build_px4fmu-v2_default/src/firmware/nuttx/firmware_nuttx...Segmentation fault: 11

This does not happen when compiling using gcc 4.8.4

LorenzMeier, 2016-01-06:
Thanks for reporting! Our experience with GCC is in general underwhelming, new versions commonly had compiler bugs. Which is why we tend to recommend "the last version", in this case 4.8.4.

davids5, 2016-01-06:
Just to ass some insight. I had this issues with 4.8.4 when trying to use more aggressive optimization settings. See https://github.com/UAVCAN/libuavcan/issues/33#issuecomment-106808534

NaterGator, 2016-04-28:
This is biting me in the ass right now. Looks like the demangler is stuck in a recursive loop on EnableIf<sizeof ..?>

https://gist.github.com/NaterGator/be5412e4ce1401031234cc73d29ed8ae

closed 2016-05-22 by LorenzMeier