PX4/PX4-Autopilot #3464: Building only POSIX SITL requires genromfs

AndreasAntener, 2016-01-08:
I tried to get posix sitl running on ubuntu without having the nuttx/arm toolchain installed. Then you'll hit the error described here:
https://groups.google.com/forum/embed/#!topic/px4users/24Q5e_pwga0

Is genromfs necessary for sitl? or can we make the build not require it for those targets?

LorenzMeier, 2016-01-08:
Its too confusing for people to install a subset of tools. I will require instead to install all of them. Making it more flexible just means people skip over setup steps, move from hardware to SITL or the other way around and complain because their install is incomplete.

closed 2016-01-08 by LorenzMeier

LorenzMeier, 2016-01-08:
Fixed here: https://github.com/PX4/Devguide/commit/b6119083cf068ab7e2a7556e615402661f1a4128