PX4/PX4-Autopilot #3474: make posix_sitl_default gazebo problem

won13y, 2016-01-09:
wkyoun@wkyoun-XPS-13-9343:~/Firmware$ make posix_sitl_default gazebo

'make -C /home/wkyoun/Firmware/build_posix_sitl_default -j4 --no-print-directory gazebo
[  2%] Built target df_driver_framework
[  3%] Built target git_matrix
[  3%] Built target git_genmsg
[  4%] Built target git_gencpp
[  4%] Built target prebuild_targets
[  4%] Built target git_mavlink
[  5%] Built target generate_topic_listener
[  6%] Built target xml_gen
[  6%] Built target git_driverframework
[  6%] Built target mixer_gen
[  7%] [  7%] Built target msg_gen
Built target git_jmavsim
[  7%] Built target git_gazebo
[  7%] Built target git_uavcan
[  7%] Built target platforms__common
[  8%] Built target drivers__boards__sitl
[  8%] Built target drivers__pwm_out_sim
[ 11%] Built target platforms__posix__px4_layer
[ 15%] Built target drivers__device
[ 15%] [ 15%] Built target platforms__posix__drivers__gpssim
Built target platforms__posix__drivers__adcsim
[ 16%] [ 23%] Built target platforms__posix__drivers__tonealrmsim
Built target platforms__posix__work_queue
[ 24%] [ 25%] Built target platforms__posix__drivers__accelsim
Built target platforms__posix__drivers__ledsim
[ 26%] Built target platforms__posix__drivers__airspeedsim
[ 27%] Built target platforms__posix__drivers__barosim
[ 27%] Built target platforms__posix__drivers__rgbledsim
[ 27%] Built target systemcmds__param
[ 27%] Built target platforms__posix__drivers__gyrosim
[ 27%] Built target systemcmds__esc_calib
[ 28%] Built target systemcmds__mixer
[ 29%] Built target systemcmds__ver
[ 29%] Built target systemcmds__reboot
[ 30%] Built target systemcmds__topic_listener
[ 31%] Built target systemcmds__perf
[ 32%] Built target modules__param
[ 32%] Built target modules__sensors
[ 37%] Built target modules__uORB
[ 40%] Built target modules__systemlib__mixer
[ 41%] Built target modules__attitude_estimator_ekf
[ 42%] Built target modules__simulator
[ 43%] Built target modules__attitude_estimator_q
[ 48%] Built target modules__mavlink
[ 49%] Built target modules__ekf_att_pos_estimator
[ 51%] [ 54%] Built target modules__position_estimator_inav
Built target modules__vtol_att_control
[ 55%] Built target modules__mc_att_control
[ 55%] Built target modules__mc_pos_control
[ 61%] Built target modules__navigator
[ 63%] Built target modules__mc_pos_control_multiplatform
[ 65%] [ 67%] [ 69%] Built target modules__fw_att_control
Built target modules__land_detector
Built target modules__mc_att_control_multiplatform
[ 70%] Built target modules__dataman
[ 71%] Built target lib__mathlib
[ 72%] Built target lib__external_lgpl
[ 75%] Built target lib__ecl
[ 77%] [ 78%] Built target modules__controllib
[ 79%] Built target modules__sdlog2
Built target lib__conversion
[ 85%] Built target modules__commander
[ 85%] [ 85%] Built target lib__mathlib__math__filter
Built target lib__geo
[ 86%] Built target lib__geo_lookup
[ 87%] Built target lib__launchdetection
[ 88%] [ 88%] Built target lib__runway_takeoff
Built target lib__terrain_estimation
[ 90%] Built target modules__fw_pos_control_l1
[ 98%] Built target modules__systemlib
[100%] Built target mainapp
SITL ARGS
rc_script: posix-configs/SITL/init/rcS
debugger: none
program: gazebo
model: none
build_path: /home/wkyoun/Firmware/build_posix_sitl_default
empty model, setting iris as default
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   thread
-- checking for module 'gazebo>=6.0'
--   package 'gazebo>=6.0' not found
CMake Error at /usr/share/cmake-3.2/Modules/FindPkgConfig.cmake:344 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.2/Modules/FindPkgConfig.cmake:506 (_pkg_check_modules_internal)
  CMakeLists.txt:7 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
See also "/home/wkyoun/Firmware/Tools/sitl_gazebo/Build/CMakeFiles/CMakeOutput.log".
make[4]: *** [src/firmware/posix/CMakeFiles/gazebo] Error 1
make[3]: *** [src/firmware/posix/CMakeFiles/gazebo.dir/all] Error 2
make[2]: *** [src/firmware/posix/CMakeFiles/gazebo.dir/rule] Error 2
make[1]: *** [gazebo] Error 2
make: *** [posix_sitl_default] Error 2'

SimonWilks, 2016-01-09:
It says: "package 'gazebo>=6.0' not found". Is it installed and recent enough? Did you read the documentation? http://dev.px4.io/simulation-gazebo.html

If you're still having problems it would be better if you ask on the mailing list as this doesn't look like a bug but an incomplete installation.

closed 2016-01-09 by SimonWilks

won13y, 2016-01-09:
@SimonWilks  Please re-open this question,

I am new to github, how can I ask this on the mailing list? 

Would you please let me know?

SimonWilks, 2016-01-09:
Sure, you can either join the mailing list or gitter chat. You can find the infos here: https://pixhawk.org/community/start