PX4/PX4-Autopilot #3475: gazebo version confict? : make posix_sitl_default gazebo problem

won13y, 2016-01-09:
When I finished installing ROS Indigo, I noticed that it already came with Gazebo(version2.2.6).

I tried to directly install the 6.0 Gazebo,

but, it was not possible because it looks like 6.0 version and 2.2.6 version conflict with each other.

Actually, I have tried to install gazebo as described in 

http://gazebosim.org/tutorials?tut=install_ubuntu&ver=6.0&cat=install

but, it says that : E: Unable to correct problems, you have held broken package

It looks like that I have Gazebo(version2.2.6) cause this problem 

(gazebo which command indicate that I have Gazebo(version2.2.6)

How can I remove Gazebo(version2.2.6) came with ROS indigo?

I don't know how to remove Gazebo(version2.2.6) and install gazebo > 6.0 in order to 

run make posix_sitl_default gazebo as described in http://dev.px4.io/simulation-gazebo.html

My situation is as follows:

wkyoun@wkyoun-XPS-13-9343:~$ sudo apt-get install gazebo6
 >
    [sudo] password for wkyoun: 
    Reading package lists... Done
    Building dependency tree       
    Reading state information... Done
    Some packages could not be installed. This may mean that you have
    requested an impossible situation or if you are using the unstable
    distribution that some required packages have not yet been created
    or been moved out of Incoming.
    The following information may help to resolve the situation:

    The following packages have unmet dependencies:
     gazebo6 : Depends: libgazebo6 (= 6.5.1-1~trusty) but it is not going to be installed
               Depends: libsdformat3 but it is not going to be installed
               Depends: gazebo6-common (= 6.5.1-1~trusty) but it is not going to be installed
               Recommends: gazebo6-plugin-base
    E: Unable to correct problems, you have held broken packages.

Also, I tried to run Gazebo as described in  

http://dev.px4.io/simulation-gazebo.html

but the error is found: 

wkyoun@wkyoun-XPS-13-9343:~/Firmware$ make posix_sitl_default gazebo

    make -C /home/wkyoun/Firmware/build_posix_sitl_default -j4 --no-print-directory gazebo
    [  2%] Built target df_driver_framework
    [  3%] Built target git_matrix
    [  3%] Built target git_genmsg
    [  4%] Built target git_gencpp
    [  4%] Built target prebuild_targets
    [  4%] Built target git_mavlink
    [  5%] Built target generate_topic_listener
    [  6%] Built target xml_gen
    [  6%] Built target git_driverframework
    [  6%] Built target mixer_gen
    [  7%] [  7%] Built target msg_gen
    Built target git_jmavsim
    [  7%] Built target git_gazebo
    [  7%] Built target git_uavcan
    [  7%] Built target platforms__common
    [  8%] Built target drivers__boards__sitl
    [  8%] Built target drivers__pwm_out_sim
    [ 11%] Built target platforms__posix__px4_layer
    [ 15%] Built target drivers__device
    [ 15%] [ 15%] Built target platforms__posix__drivers__gpssim
    Built target platforms__posix__drivers__adcsim
    [ 16%] [ 23%] Built target platforms__posix__drivers__tonealrmsim
    Built target platforms__posix__work_queue
    [ 24%] [ 25%] Built target platforms__posix__drivers__accelsim
    Built target platforms__posix__drivers__ledsim
    [ 26%] Built target platforms__posix__drivers__airspeedsim
    [ 27%] Built target platforms__posix__drivers__barosim
    [ 27%] Built target platforms__posix__drivers__rgbledsim
    [ 27%] Built target systemcmds__param
    [ 27%] Built target platforms__posix__drivers__gyrosim
    [ 27%] Built target systemcmds__esc_calib
    [ 28%] Built target systemcmds__mixer
    [ 29%] Built target systemcmds__ver
    [ 29%] Built target systemcmds__reboot
    [ 30%] Built target systemcmds__topic_listener
    [ 31%] Built target systemcmds__perf
    [ 32%] Built target modules__param
    [ 32%] Built target modules__sensors
    [ 37%] Built target modules__uORB
    [ 40%] Built target modules__systemlib__mixer
    [ 41%] Built target modules__attitude_estimator_ekf
    [ 42%] Built target modules__simulator
    [ 43%] Built target modules__attitude_estimator_q
    [ 48%] Built target modules__mavlink
    [ 49%] Built target modules__ekf_att_pos_estimator
    [ 51%] [ 54%] Built target modules__position_estimator_inav
    Built target modules__vtol_att_control
    [ 55%] Built target modules__mc_att_control
    [ 55%] Built target modules__mc_pos_control
    [ 61%] Built target modules__navigator
    [ 63%] Built target modules__mc_pos_control_multiplatform
    [ 65%] [ 67%] [ 69%] Built target modules__fw_att_control
    Built target modules__land_detector
    Built target modules__mc_att_control_multiplatform
    [ 70%] Built target modules__dataman
    [ 71%] Built target lib__mathlib
    [ 72%] Built target lib__external_lgpl
    [ 75%] Built target lib__ecl
    [ 77%] [ 78%] Built target modules__controllib
    [ 79%] Built target modules__sdlog2
    Built target lib__conversion
    [ 85%] Built target modules__commander
    [ 85%] [ 85%] Built target lib__mathlib__math__filter
    Built target lib__geo
    [ 86%] Built target lib__geo_lookup
    [ 87%] Built target lib__launchdetection
    [ 88%] [ 88%] Built target lib__runway_takeoff
    Built target lib__terrain_estimation
    [ 90%] Built target modules__fw_pos_control_l1
    [ 98%] Built target modules__systemlib
    [100%] Built target mainapp
    SITL ARGS
    rc_script: posix-configs/SITL/init/rcS
    debugger: none
    program: gazebo
    model: none
    build_path: /home/wkyoun/Firmware/build_posix_sitl_default
    empty model, setting iris as default
    -- Boost version: 1.54.0
    -- Found the following Boost libraries:
    --   system
    --   thread
    -- checking for module 'gazebo>=6.0'
    --   package 'gazebo>=6.0' not found
    CMake Error at /usr/share/cmake-3.2/Modules/FindPkgConfig.cmake:344 (message):
      A required package was not found
    Call Stack (most recent call first):
      /usr/share/cmake-3.2/Modules/FindPkgConfig.cmake:506 (_pkg_check_modules_internal)
      CMakeLists.txt:7 (pkg_check_modules)

    -- Configuring incomplete, errors occurred!
    See also "/home/wkyoun/Firmware/Tools/sitl_gazebo/Build/CMakeFiles/CMakeOutput.log".
    make[4]: *** [src/firmware/posix/CMakeFiles/gazebo] Error 1
    make[3]: *** [src/firmware/posix/CMakeFiles/gazebo.dir/all] Error 2
    make[2]: *** [src/firmware/posix/CMakeFiles/gazebo.dir/rule] Error 2
    make[1]: *** [gazebo] Error 2
    make: *** [posix_sitl_default] Error 2

devbharat, 2016-01-09:
I think you can solve this by removing all ros-gazebo* packages with apt-get remove. Then, install gazebo6 as the gazebosim documentation says, and then build other remaining ros-gazebo packages (like gazebo_msgs) in a catkin workspace from source.

won13y, 2016-01-11:
@devbharat  
Thank you for your reply

It was solved, but still another error is found as following:

I don't know why: please let me know?

**wkyoun@wkyoun-XPS-13-9343:~/Firmware$ make posix_sitl_default gazebo**

     make -C /home/wkyoun/Firmware/build_posix_sitl_default -j4 --no-print-directory gazebo
     [  2%] Built target df_driver_framework
     [  2%] [  3%] Built target git_genmsg
     Built target git_gencpp
     [  4%] Built target git_matrix
     [  4%] [  4%] Built target prebuild_targets
     Built target git_mavlink
     [  5%] Built target generate_topic_listener
     [  6%] Built target xml_gen
     [  6%] Built target git_driverframework
     [  6%] Built target mixer_gen
     [  7%] [  7%] Built target msg_gen
     Built target git_jmavsim
     [  7%] [  7%] Built target git_gazebo
     Built target git_uavcan
     [  7%] Built target platforms__common
     [  7%] [  8%] Built target drivers__pwm_out_sim
     Built target drivers__boards__sitl
     [ 11%] [ 15%] Built target platforms__posix__px4_layer
     Built target drivers__device
     [ 15%] Built target platforms__posix__drivers__adcsim
     [ 16%] [ 23%] Built target platforms__posix__drivers__tonealrmsim
     Built target platforms__posix__work_queue
     [ 23%] Built target platforms__posix__drivers__gpssim
     [ 25%] [ 25%] [ 26%] Built target platforms__posix__drivers__barosim
     Built target platforms__posix__drivers__accelsim
     Built target platforms__posix__drivers__airspeedsim
     [ 27%] Built target platforms__posix__drivers__ledsim
     [ 27%] [ 27%] [ 27%] Built target platforms__posix__drivers__gyrosim
     Built target systemcmds__param
     Built target platforms__posix__drivers__rgbledsim
     [ 27%] Built target systemcmds__esc_calib
     [ 28%] Built target systemcmds__mixer
     [ 28%] Built target systemcmds__reboot
     [ 29%] [ 30%] Built target systemcmds__ver
     Built target systemcmds__topic_listener
     [ 31%] Built target systemcmds__perf
     [ 32%] Built target modules__param
     [ 32%] Built target modules__sensors
     [ 37%] Built target modules__uORB
     [ 40%] Built target modules__systemlib__mixer
     [ 41%] Built target modules__attitude_estimator_ekf
     [ 42%] Built target modules__attitude_estimator_q
     [ 43%] Built target modules__simulator
     [ 44%] [ 49%] Built target modules__ekf_att_pos_estimator
     Built target modules__mavlink
     [ 51%] Built target modules__position_estimator_inav
     [ 54%] Built target modules__vtol_att_control
     [ 55%] Built target modules__mc_att_control
     [ 55%] Built target modules__mc_pos_control
     [ 61%] Built target modules__navigator
     [ 63%] Built target modules__mc_pos_control_multiplatform
     [ 65%] Built target modules__land_detector
     [ 67%] [ 69%] [ 70%] Built target modules__fw_att_control
     Built target modules__dataman
     Built target modules__mc_att_control_multiplatform
     [ 71%] Built target lib__external_lgpl
     [ 72%] Built target lib__mathlib
     [ 75%] [ 76%] Built target modules__sdlog2
     Built target lib__ecl
     [ 78%] Built target modules__controllib
     [ 79%] Built target lib__conversion
     [ 79%] Built target lib__mathlib__math__filter
     [ 85%] Built target modules__commander
     [ 86%] Built target lib__geo_lookup
     [ 86%] Built target lib__geo
     [ 87%] Built target lib__launchdetection
     [ 87%] [ 88%] Built target lib__terrain_estimation
     Built target lib__runway_takeoff
     [ 96%] Built target modules__systemlib
     [ 98%] Built target modules__fw_pos_control_l1
     [100%] Built target mainapp
     SITL ARGS
     rc_script: posix-configs/SITL/init/rcS
     debugger: none
     program: gazebo
     model: none
     build_path: /home/wkyoun/Firmware/build_posix_sitl_default
     empty model, setting iris as default
     -- Boost version: 1.54.0
     -- Found the following Boost libraries:
     --   system
     --   thread
     -- Configuring done
     -- Generating done
     -- Build files have been written to: /home/wkyoun/Firmware/Tools/sitl_gazebo/Build
     make[5]: warning: -jN forced in submake: disabling jobserver mode.
     make[7]: *** No rule to make target `../msgs/PROTOBUF_PROTOC_EXECUTABLE-NOTFOUND', needed by `msgs/MotorSpeed.pb.cc'.  Stop.
     make[6]: *** [msgs/CMakeFiles/mav_msgs.dir/all] Error 2
     make[6]: *** Waiting for unfinished jobs....
     [  3%] Built target hello_world
     make[5]: *** [all] Error 2
     make[4]: *** [src/firmware/posix/CMakeFiles/gazebo] Error 2
     make[3]: *** [src/firmware/posix/CMakeFiles/gazebo.dir/all] Error 2
     make[2]: *** [src/firmware/posix/CMakeFiles/gazebo.dir/rule] Error 2
     make[1]: *** [gazebo] Error 2
     make: *** [posix_sitl_default] Error 2

sanderux, 2016-01-11:
on ubuntu
install libogre6-dev
Change the CmakeList.txt to add include path /usr/lib/OGRE

won13y, 2016-01-12:
@sanderux 
Thank you for your help.
But, I was not able to install libgazebo6-dev by commanding "sudo apt-get install libgazebo6-dev "

And commanding "apt-cache search libgazebo6-dev" results in nothing.

Would you please let me know how to install libgazebo6-dev?

Last thing,

How can you figure out from error result that   libgazebo6-dev is the root cause of the error?
It seems to me that there are not any indications related to  libgazebo6.

sanderux, 2016-01-12:
http://packages.ubuntu.com/libgazebo6-dev is there, check your repo's
the OGRE path is not documented, you wil find that problem when you passed this one

julianoes, 2016-01-26:
I believe I had to install this on Ubuntu 15.10:

```
sudo apt-get install libogre-1.9-dev
```

sanderux, 2016-01-26:
the include path should be added here:
https://github.com/PX4/sitl_gazebo/blob/master/CMakeLists.txt#L23

closed 2016-05-22 by LorenzMeier

marianobianchi, 2016-06-07:
I had the same problem. I could run "make posix_sitl_default gazebo" when included "find_package(Protobuf REQUIRED)" in line 24 from https://github.com/PX4/sitl_gazebo/blob/master/CMakeLists.txt.

I''m running Ubuntu 16.04 and ROS Kinetic.

Why is this issue closed if it isn't already solved?

vinhk, 2016-06-16:
@won13y Were you able to have that issue fix? I am having the same issue. Please help