PX4/PX4-Autopilot #3487: PX4 on STM32F4Discovery

nopeppermint, 2016-01-10:
I like to port the PX4 on STM32F4Discovery manual 
https://pixhawk.org/modules/stm32f4discovery#flashing_the_px4_bootloader
to http://dev.px4.io/
But therefore I like to get this thing again working ;-)

I am trying to get the STM32f4Discovery working with the actual bootloader/firmware
I am working on linux.
Bootloader is builded successfully, and flashed via stlink (red led is flashing after each reset)
Firmware is builded successfully with ``make px4-stm32f4discovery_default``
but with the additional ``update`` to load the firmware with the bootloader doesn't work:
``[ 98%] uploading /home/stefan/src/firmware/build_px4-stm32f4discovery_default/src/firmware/nuttx/nuttx-px4-stm32f4discovery-default.px4
Loaded firmware for 63,0, size: 350212 bytes, waiting for the bootloader...
If the board does not respond within 1-2 seconds, unplug and re-plug the USB connector``
=> here it stops, even If I press the black Reset Button and the red led starts flashing

The Board is connected over the two USB Cables

The bootloader usb port is visible as /dev/ttyACM0 and the stlink as /dev/ttyUSB0

So I tried also to flash the firmware with:
``firmware/Tools $ python ./px_uploader.py --port /dev/ttyACM0 ../build_px4-stm32f4discovery_default/src/firmware/nuttx/nuttx-px4-stm32f4discovery-default.px4 
Loaded firmware for 63,0, size: 350212 bytes, waiting for the bootloader...
If the board does not respond within 1-2 seconds, unplug and re-plug the USB connector.``
=> here it stops, even If I press the black Reset Button and the red led starts flashing

The stlink:
``[ 6119.725240] usb 1-2: new full-speed USB device number 58 using xhci_hcd
[ 6119.910477] usb 1-2: New USB device found, idVendor=0483, idProduct=3748
[ 6119.910486] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3``

The bootloader usb port:
``[ 6236.379183] usb 1-1: new full-speed USB device number 62 using xhci_hcd
[ 6236.564315] usb 1-1: New USB device found, idVendor=26ac, idProduct=0001
[ 6236.564331] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 6236.565453] cdc_acm 1-1:1.0: ttyACM0: USB ACM device``

LorenzMeier, 2016-01-10:
Try to read the Tools/px_uploader.py script and see if it locks out the Disco board through PID / VID or something else.

Can I have the output of lsusb with verbose flag?

nopeppermint, 2016-01-10:
It was a permission problem to acces /dev/ttyACM0  ;-)

nopeppermint, 2016-01-10:
firmware/Tools $ sudo python ./px_uploader.py --port /dev/ttyACM0 ../build_px4-stm32f4discovery_default/src/firmware/nuttx/nuttx-px4-stm32f4discovery-default.px4 
Loaded firmware for 63,0, size: 350212 bytes, waiting for the bootloader...
If the board does not respond within 1-2 seconds, unplug and re-plug the USB connector.
attempting reboot on /dev/ttyACM0...
if the board does not respond, unplug and re-plug the USB connector.
attempting reboot on /dev/ttyACM0...
if the board does not respond, unplug and re-plug the USB connector.
Found board 63,0 bootloader rev 5 on /dev/ttyACM0
ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff type: ÿÿÿÿ
idtype: =FF
vid: ffffffff
pid: ffffffff
coa: //////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////8=

sn: 002200223231471432383435
chip: 10016413
family: STM32F40x
revision: Z
flash 1032192

Erase  : [====================] 100.0%
Program: [====================] 100.0%
Verify : [====================] 100.0%
Rebooting.

closed 2016-01-10 by LorenzMeier