PX4/PX4-Autopilot #3490: Revisit git submodule magic

julianoes, 2016-01-10:
We need to look into the git submodule magic because there are some corner cases where data can get lost. One of the cases is when files are added in a submodule but not committed.

Example steps to replicate:
```
echo FOO > Tools/jMAVSim/new_file.txt
(cd Tools/jMAVSim && git add new_file.txt)
make clean
make
```

closed 2016-01-11 by LorenzMeier

julianoes, 2016-01-11:
Great, thanks!

FYI: @georgehines

dagar, 2016-01-11:
This is one the problems with submodules where there might not be a good solution. I think we're going to find that running `git submodule update --init` before cmake fixed a lot of problems with people not cloning recursively or having out of date submodules.

Having `make clean` recurse into each build directory and run clean might be what you really want instead of the current behaviour.

jgoppert, 2016-01-12:
CMake is not designed to have external libraries living in the source tree, specifically, it is not appropriate to do a add_subdirectory(external_library)

All external libraries should be built how I built nuttx, in the build directory. Then linked to at compile time. This is standard cmake procedure, and  you can think of it as similar to a ros dev workspace.

See the mechanics here: https://cmake.org/cmake/help/v3.0/module/ExternalProject.html

jgoppert, 2016-01-12:
For developers who want to edit  git submodules within the source of qgroundcontrol, this might be another option: http://www.diracprogram.org/doc/release-12/programmers/external_projects.html, although I have never tried it yet.

commit f99d0525 (2016-01-11): Fix submodule force. Fixes #3490.