PX4/PX4-Autopilot #3492: VTOL: Support MAVLink commands for direct VTOL takeoff/land

AndreasAntener, 2016-01-11:
https://github.com/mavlink/mavlink/pull/487

sanderux, 2016-01-11:
Should these commands be supported in FW mode?
and if so, should they be interpreted as
MAV_CMD_NAV_VTOL_TAKEOFF: if mode=FW then Transition - TakeOff - Transition
MAV_CMD_NAV_VTOL_LAND:        if mode=FW then Transition - Land - Transition

sanderux, 2016-02-04:
For my own reference:
Treat as transition command with:
https://github.com/PX4/Firmware/blob/9a8e94bb68d2c78ea1bd9d62ac993d937b797a31/src/modules/navigator/mission.cpp#L424

AndreasAntener, 2016-02-08:
Why would you transition to FW for takeoff? It's a VTOL ;)

sanderux, 2016-02-08:
Actualy it transitions after takeoff, so it climbs to takeoff height as vtol and then transitions to fw to travel to the takeoff waypoint. try in sitl, it works nicely.

sanderux, 2016-02-08:
@AndreasAntener @LorenzMeier @tumbili 
Considering we have RTL transitioning, takeoff is basically always vtol, and ada's valid point "it's a vtol". 
Would it be more user friendly to have LAND be VTOL by default, controlled by a param). This would make switching to LAND mode a failsafe feature and would eliminate the need for a revision of all other failsafe procedures (GPS loss, engine failure, etc).

AndreasAntener, 2016-04-04:
Implemented and merged

closed 2016-04-04 by AndreasAntener