PX4/PX4-Autopilot #3493: HIL spits out continues stream of warnings

DonLakeFlyer, 2016-01-11:
- Using latest firmware master and master QGC
- Boot QGC and connect for an XPlane HIL session
- Connect to HIL using the widget

Firmware spits out continuous stream of:
WARN  reset: NaN in states, resetting
telem> [ekf] ref: LA 0.0000,LO 0.0000,ALT 0.00
WARN  reset: NaN in states, resetting
telem> [ekf] ref: LA 0.0000,LO 0.0000,ALT 0.00
WARN  navigator timed out
WARN  reset: NaN in states, resetting

jmullen, 2016-01-16:
Did you have a resolution to this error? I'm having the same issue, latest QQC, using XPlane 10 HIL session, I get a vomit of errors saying:

Critical: [ekf check] NaN in states, resetting
Info: [ekf] ref: LA 32.2043,LO -100.0420,ALT 743.12

No amount of googling has come up with a resolution to this issue.

DonLakeFlyer, 2016-01-16:
I figured Lorenz fixed it since he closed the Issue

jmullen, 2016-01-17:
The issue is still persisting after pulling the head.

LorenzMeier, 2016-01-17:
@jmullen I have looked into it but its not fixed yet. QGC is saturating the link.

LorenzMeier, 2016-01-17:
Doing the testing now..

nsblackwell, 2016-04-24:
I'm running into this as well -- any updates?

LorenzMeier, 2016-12-26:
Closing as stale.

closed 2016-12-26 by LorenzMeier