PX4/PX4-Autopilot #3495: Running simulation Gazebo/jMavsim fail

harewei, 2016-01-12:
Hi All

I have installed both jmavsim and gazebo as instructed in 
https://pixhawk.org/dev/hil/jmavsim
http://gazebosim.org/tutorials?tut=install_ubuntu&ver=6.0&cat=install

I also have SITL working from
http://dev.ardupilot.com/wiki/setting-up-sitl-on-linux/

But when I retrieved from PX4/Firmware
https://github.com/PX4/Firmware
and try to make with either make posix_sitl_default jmavsim or make posix_sitl_default gazebo (http://dev.px4.io/simulation-gazebo.html)(http://dev.px4.io/simulation-sitl.html), it gives me errors saying either
make -C /home/harewei/Firmware/build_posix_sitl_default -j4 --no-print-directory jmavsim
make[1]: *** No rule to make target `jmavsim'.  Stop.
make: *** [posix_sitl_default] Error 2

make -C /home/harewei/Firmware/build_posix_sitl_default -j4 --no-print-directory gazebo
make[1]: *** No rule to make target `gazebo'.  Stop.
make: *** [posix_sitl_default] Error 2

what am I doing wrong?

LorenzMeier, 2016-01-12:
You're not using the right make commands. Use the ones in the documentation, not something else. And provide the complete log including the commands you've entered.

harewei, 2016-01-12:
I'm not sure what you mean?
all my commands and output there in my post.  also, I followed the exact command given in the developer guide shown in the links.
just to make it clear

running 
*make posix_sitl_default jmavsim
gave me
make -C /home/harewei/Firmware/build_posix_sitl_default -j4 --no-print-directory jmavsim
make[1]: *** No rule to make target `jmavsim'. Stop.
make: *** [posix_sitl_default] Error 2

running
*make posix_sitl_default gazebo
gave me
make -C /home/harewei/Firmware/build_posix_sitl_default -j4 --no-print-directory gazebo
make[1]: *** No rule to make target `gazebo'. Stop.
make: *** [posix_sitl_default] Error 2

I know the commands are right because I managed to get jmavsim working before my reformat (not gazebo though).  but I can't figure out what's wrong this time

LorenzMeier, 2016-01-12:
Try to pull from master and try again please.

harewei, 2016-01-13:
I tried it again with command
~/Firmware$ make posix_sitl_default gazebo

gave me
git submodule init
Submodule 'NuttX' (git://github.com/PX4/NuttX.git) registered for path 'NuttX'
Submodule 'Tools/gencpp' (https://github.com/ros/gencpp.git) registered for path 'Tools/gencpp'
Submodule 'Tools/genmsg' (https://github.com/ros/genmsg.git) registered for path 'Tools/genmsg'
Submodule 'Tools/jMAVSim' (https://github.com/PX4/jMAVSim.git) registered for path 'Tools/jMAVSim'
Submodule 'Tools/sitl_gazebo' (https://github.com/PX4/sitl_gazebo.git) registered for path 'Tools/sitl_gazebo'
Submodule 'cmake/cmake_hexagon' (https://github.com/ATLFlight/cmake_hexagon) registered for path 'cmake/cmake_hexagon'
Submodule 'mavlink/include/mavlink/v1.0' (git://github.com/mavlink/c_library.git) registered for path 'mavlink/include/mavlink/v1.0'
Submodule 'src/lib/DriverFramework' (https://github.com/PX4/DriverFramework.git) registered for path 'src/lib/DriverFramework'
Submodule 'src/lib/dspal' (https://github.com/ATLFlight/dspal.git) registered for path 'src/lib/dspal'
Submodule 'src/lib/ecl' (https://github.com/PX4/ecl.git) registered for path 'src/lib/ecl'
Submodule 'src/lib/matrix' (https://github.com/PX4/Matrix.git) registered for path 'src/lib/matrix'
Submodule 'src/modules/uavcan/libuavcan' (git://github.com/UAVCAN/libuavcan.git) registered for path 'src/modules/uavcan/libuavcan'
Submodule 'unittests/googletest' (https://github.com/google/googletest.git) registered for path 'unittests/googletest'
Tools/check_submodules.sh 
Checked NuttX submodule, correct version found
Checked Tools/gencpp submodule, correct version found
Checked Tools/genmsg submodule, correct version found
Checked Tools/jMAVSim submodule, correct version found
Checked Tools/sitl_gazebo submodule, correct version found
Checked cmake/cmake_hexagon submodule, correct version found
Checked mavlink/include/mavlink/v1.0 submodule, correct version found
Checked src/lib/DriverFramework submodule, correct version found
Checked src/lib/dspal submodule, correct version found
Checked src/lib/ecl submodule, correct version found
Checked src/lib/matrix submodule, correct version found
Checked src/modules/uavcan/libuavcan submodule, correct version found
Checked unittests/googletest submodule, correct version found
Synchronizing submodule url for 'NuttX'
Synchronizing submodule url for 'Tools/gencpp'
Synchronizing submodule url for 'Tools/genmsg'
Synchronizing submodule url for 'Tools/jMAVSim'
Synchronizing submodule url for 'Tools/sitl_gazebo'
Synchronizing submodule url for 'cmake/cmake_hexagon'
Synchronizing submodule url for 'mavlink/include/mavlink/v1.0'
Synchronizing submodule url for 'src/lib/DriverFramework'
Synchronizing submodule url for 'src/lib/dspal'
Synchronizing submodule url for 'src/lib/ecl'
Synchronizing submodule url for 'src/lib/matrix'
Synchronizing submodule url for 'src/modules/uavcan/libuavcan'
Synchronizing submodule url for 'unittests/googletest'
-- posix-sitl-default
-- The CXX compiler identification is GNU 4.8.4
-- The C compiler identification is GNU 4.8.4
-- The ASM compiler identification is GNU
-- Found assembler: /usr/lib/ccache/cc
-- Check for working CXX compiler: /usr/lib/ccache/c++
-- Check for working CXX compiler: /usr/lib/ccache/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/lib/ccache/cc
-- Check for working C compiler: /usr/lib/ccache/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PythonInterp: /usr/bin/python (found version "2.7.6") 
CMake Error at CMakeLists.txt:332 (add_subdirectory):
  The source directory

    /home/harewei/Firmware/src/lib/ecl

  does not contain a CMakeLists.txt file.

CMake Warning (dev) at cmake/posix/px4_impl_posix.cmake:91 (get_target_property):
  Policy CMP0045 is not set: Error on non-existent target in
  get_target_property.  Run "cmake --help-policy CMP0045" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  get_target_property() called with non-existent target "lib__ecl".
Call Stack (most recent call first):
  src/firmware/posix/CMakeLists.txt:4 (px4_posix_generate_builtin_commands)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at cmake/posix/px4_impl_posix.cmake:91 (get_target_property):
  Policy CMP0045 is not set: Error on non-existent target in
  get_target_property.  Run "cmake --help-policy CMP0045" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  get_target_property() called with non-existent target "lib__ecl".
Call Stack (most recent call first):
  src/firmware/posix/CMakeLists.txt:4 (px4_posix_generate_builtin_commands)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at cmake/posix/px4_impl_posix.cmake:91 (get_target_property):
  Policy CMP0045 is not set: Error on non-existent target in
  get_target_property.  Run "cmake --help-policy CMP0045" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  get_target_property() called with non-existent target "lib__ecl".
Call Stack (most recent call first):
  src/firmware/posix/CMakeLists.txt:4 (px4_posix_generate_builtin_commands)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at CMakeLists.txt:340 (add_subdirectory):
  add_subdirectory given source "src/lib/DriverFramework/framework/src" which
  is not an existing directory.

-- GIT_DESC = 181cbd383cf51ac6e2e7f55151461c0fb9c97001
-- Configuring incomplete, errors occurred!
See also "/home/harewei/Firmware/build_posix_sitl_default/CMakeFiles/CMakeOutput.log".
make: *** [posix_sitl_default] Error 1

harewei, 2016-01-13:
Making a second post to display a second approach
I moved the Firmware folder under my ros catkin workspace
~catkin_ws/src
and used catkin_make

The results before the build failed
Generating C++ code from px4/actuator_controls.msg
[ 80%] Generating C++ code from px4/control_state.msg
[ 81%] Generating C++ code from px4/vehicle_status.msg
[ 81%] Generating C++ code from px4/vehicle_local_position_setpoint.msg
[ 81%] Generating C++ code from px4/vehicle_local_position.msg
[ 82%] Generating C++ code from px4/mc_virtual_rates_setpoint.msg
[ 82%] Built target copter_visualization
[ 82%] Built target gcs_image_bridge
[ 82%] Generating C++ code from px4/position_setpoint.msg
[ 82%] Generating C++ code from px4/rc_channels.msg
[ 85%] Built target mavros_extras
[ 86%] [ 92%] Built target mavros_plugins
Generating C++ code from px4/manual_control_setpoint.msg
[ 92%] Built target sitl_test
[ 92%] Generating C++ code from px4/vehicle_control_mode.msg
[ 93%] Built target sitl_test_node
[ 93%] Generating C++ code from px4/vehicle_global_velocity_setpoint.msg
[ 93%] Built target px4_generate_messages_cpp
[ 93%] Built target px4_generate_messages
Scanning dependencies of target px4
[ 94%] [ 94%] [ 94%] [ 94%] Building CXX object Firmware/CMakeFiles/px4.dir/src/platforms/ros/perf_counter.cpp.o
Building CXX object Firmware/CMakeFiles/px4.dir/src/platforms/ros/px4_ros_impl.cpp.o
Building CXX object Firmware/CMakeFiles/px4.dir/src/platforms/ros/geo.cpp.o
Building CXX object Firmware/CMakeFiles/px4.dir/src/lib/mathlib/math/Limits.cpp.o
/home/harewei/catkin_ws/src/Firmware/src/lib/mathlib/math/Limits.cpp:51:0: warning: "M_PI_F" redefined [enabled by default]
 #define M_PI_F 3.14159265358979323846f
 ^
In file included from /home/harewei/catkin_ws/src/Firmware/src/lib/mathlib/math/Limits.hpp:42:0,
                 from /home/harewei/catkin_ws/src/Firmware/src/lib/mathlib/math/Limits.cpp:44:
/home/harewei/catkin_ws/src/Firmware/src/platforms/px4_defines.h:194:0: note: this is the location of the previous definition
 #define M_PI_F   (float)M_PI
 ^
[ 95%] Building CXX object Firmware/CMakeFiles/px4.dir/src/modules/systemlib/circuit_breaker.cpp.o
Linking CXX shared library /home/harewei/catkin_ws/devel/lib/libpx4.so
[ 95%] Built target px4
Scanning dependencies of target attitude_estimator
Scanning dependencies of target demo_offboard_attitude_setpoints
Scanning dependencies of target commander
[ 95%] Building CXX object Firmware/CMakeFiles/attitude_estimator.dir/src/platforms/ros/nodes/attitude_estimator/attitude_estimator.cpp.o
Scanning dependencies of target demo_offboard_position_setpoints
[ 96%] [ 96%] Building CXX object Firmware/CMakeFiles/commander.dir/src/platforms/ros/nodes/commander/commander.cpp.o
Building CXX object Firmware/CMakeFiles/demo_offboard_attitude_setpoints.dir/src/platforms/ros/nodes/demo_offboard_attitude_setpoints/demo_offboard_attitude_setpoints.cpp.o
[ 97%] Building CXX object Firmware/CMakeFiles/demo_offboard_position_setpoints.dir/src/platforms/ros/nodes/demo_offboard_position_setpoints/demo_offboard_position_setpoints.cpp.o
In file included from /home/harewei/catkin_ws/src/Firmware/src/lib/mathlib/mathlib.h:45:0,
                 from /home/harewei/catkin_ws/src/Firmware/src/platforms/ros/nodes/attitude_estimator/attitude_estimator.cpp:44:
/home/harewei/catkin_ws/src/Firmware/src/lib/mathlib/math/Matrix.hpp:52:27: fatal error: matrix/math.hpp: No such file or directory
 #include "matrix/math.hpp"
                           ^
compilation terminated.
Linking CXX executable /home/harewei/catkin_ws/devel/lib/px4/demo_offboard_position_setpoints
[ 97%] Built target demo_offboard_position_setpoints
Scanning dependencies of target manual_input
[ 97%] Building CXX object Firmware/CMakeFiles/manual_input.dir/src/platforms/ros/nodes/manual_input/manual_input.cpp.o
Linking CXX executable /home/harewei/catkin_ws/devel/lib/px4/demo_offboard_attitude_setpoints
[ 97%] Built target demo_offboard_attitude_setpoints
Scanning dependencies of target mavlink
[ 97%] Building CXX object Firmware/CMakeFiles/mavlink.dir/src/platforms/ros/nodes/mavlink/mavlink.cpp.o
Linking CXX executable /home/harewei/catkin_ws/devel/lib/px4/commander
[ 97%] Built target commander
Scanning dependencies of target mc_att_control
[ 97%] Building CXX object Firmware/CMakeFiles/mc_att_control.dir/src/modules/mc_att_control_multiplatform/mc_att_control_main.cpp.o
In file included from /home/harewei/catkin_ws/src/Firmware/src/lib/mathlib/mathlib.h:45:0,
                 from /home/harewei/catkin_ws/src/Firmware/src/modules/mc_att_control_multiplatform/mc_att_control.h:63,
                 from /home/harewei/catkin_ws/src/Firmware/src/modules/mc_att_control_multiplatform/mc_att_control_main.cpp:55:
[... 9 lines cut ...]
[ 97%] Built target manual_input
[ 98%] Building CXX object Firmware/CMakeFiles/mc_att_control.dir/src/modules/mc_att_control_multiplatform/mc_att_control_base.cpp.o
In file included from /home/harewei/catkin_ws/src/Firmware/src/lib/mathlib/mathlib.h:45:0,
                 from /home/harewei/catkin_ws/src/Firmware/src/modules/mc_att_control_multiplatform/mc_att_control.h:63,
                 from /home/harewei/catkin_ws/src/Firmware/src/modules/mc_att_control_multiplatform/mc_att_control.cpp:46:
/home/harewei/catkin_ws/src/Firmware/src/lib/mathlib/math/Matrix.hpp:52:27: fatal error: matrix/math.hpp: No such file or directory
 #include "matrix/math.hpp"
                           ^
compilation terminated.
[... cut ...]
make[2]: *** [Firmware/CMakeFiles/mc_att_control.dir/src/modules/mc_att_control_multiplatform/mc_att_control_main.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 98%] Built target mavlink
make[2]: *** [Firmware/CMakeFiles/mc_att_control.dir/src/modules/mc_att_control_multiplatform/mc_att_control.cpp.o] Error 1
make[2]: *** [Firmware/CMakeFiles/mc_att_control.dir/src/modules/mc_att_control_multiplatform/mc_att_control_base.cpp.o] Error 1
make[1]: *** [Firmware/CMakeFiles/mc_att_control.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

harewei, 2016-01-14:
OK, someone must have fixed the problem today, because running the exact same command after git clone a fresh copy works now for jamvsim.  If I run gazebo it still causes problem though.

harewei@AWCC:~/Firmware$ make posix_sitl_default gazebo
git submodule init
Tools/check_submodules.sh 
Git submodule config valid.
Checked NuttX submodule, correct version found
Checked Tools/gencpp submodule, correct version found
Checked Tools/genmsg submodule, correct version found
Checked Tools/jMAVSim submodule, correct version found
Checked Tools/sitl_gazebo submodule, correct version found
Checked cmake/cmake_hexagon submodule, correct version found
Checked mavlink/include/mavlink/v1.0 submodule, correct version found
Checked src/lib/DriverFramework submodule, correct version found
Checked src/lib/dspal submodule, correct version found
Checked src/lib/ecl submodule, correct version found
Checked src/lib/matrix submodule, correct version found
Checked src/modules/uavcan/libuavcan submodule, correct version found
Checked unittests/googletest submodule, correct version found
make -C /home/harewei/Firmware/build_posix_sitl_default -j4 --no-print-directory gazebo
[  2%] Built target df_driver_framework
[  2%] Built target git_gencpp
[  3%] Built target git_genmsg
[  4%] Built target git_matrix
[  4%] [  4%] Built target prebuild_targets
Built target git_mavlink
[  5%] Built target generate_topic_listener
[  6%] Built target xml_gen
[  6%] Built target mixer_gen
[  6%] Built target git_driverframework
[  6%] Built target git_jmavsim
[  6%] Built target git_uavcan
[  7%] [  8%] Built target msg_gen
Built target git_gazebo
[  8%] [  8%] Built target git_ecl
Built target platforms__common
[  9%] Built target drivers__boards__sitl
[ 10%] [ 12%] Built target drivers__pwm_out_sim
Built target platforms__posix__px4_layer
[ 15%] Built target drivers__device
[ 15%] Built target platforms__posix__drivers__adcsim
[ 16%] Built target platforms__posix__drivers__gpssim
[ 23%] Built target platforms__posix__work_queue
[ 24%] Built target platforms__posix__drivers__tonealrmsim
[ 25%] Built target platforms__posix__drivers__accelsim
[ 26%] Built target platforms__posix__drivers__airspeedsim
[ 26%] Built target platforms__posix__drivers__gyrosim
[ 26%] Built target platforms__posix__drivers__rgbledsim
[ 26%] [ 27%] Built target platforms__posix__drivers__ledsim
Built target platforms__posix__drivers__barosim
[ 27%] Built target systemcmds__esc_calib
[ 27%] Built target systemcmds__param
[ 27%] [ 28%] [ 29%] Built target systemcmds__reboot
Built target systemcmds__ver
Built target systemcmds__mixer
[ 30%] [ 31%] Built target modules__param
Built target systemcmds__perf
[ 32%] [ 36%] Built target systemcmds__topic_listener
Built target modules__uORB
[ 36%] Built target modules__mc_pos_control
[ 36%] [ 39%] Built target modules__sensors
Built target modules__systemlib__mixer
[ 46%] Built target modules__systemlib
[ 47%] Built target modules__simulator
[ 48%] Built target modules__attitude_estimator_ekf
[ 50%] Built target modules__attitude_estimator_q
[ 52%] Built target modules__ekf_att_pos_estimator
[ 54%] Built target modules__position_estimator_inav
[ 59%] Built target modules__mavlink
[ 65%] [ 66%] [ 69%] Built target modules__navigator
[... 55 lines cut ...]

Ready to fly.

INFO  LED::init
INFO  LED::init
INFO  Not using /dev/ttyACM0 for radio control input. Assuming joystick input via MAVLink.
INFO  Waiting for initial data on UDP. Please start the flight simulator to proceed..
Gazebo multi-robot simulator, version 6.5.1
Copyright (C) 2012-2015 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 172.20.10.3
Gazebo multi-robot simulator, version 6.5.1
Copyright (C) 2012-2015 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Err] [Master.cc:52] EXCEPTION: Unable to start server[bind: Address already in use]. There is probably another Gazebo process running.

[Err] [Master.cc:52] EXCEPTION: Unable to start server[bind: Address already in use]. There is probably another Gazebo process running.

[Err] [InsertModelWidget.cc:298] Missing model.config for model "/home/harewei/Firmware/Tools/sitl_gazebo/models/rotors_description"

LorenzMeier, 2016-01-14:
Building with Catkin is not supported. Please stick to the instructions on http://dev.px4.io.

closed 2016-01-14 by LorenzMeier