PX4/PX4-Autopilot #3501: Standard VTOL issues

tumbili, 2016-01-13:
This is a list of all the issues of the standard VTOL code (many issues will be also applicable to the other VTOL types). The new simulation #3500 will allow to find and fix many of those issues.

 - [ ] Doing a transition in mission: In the simulation the plane starts to pitch up alot during the transition because the pitch setpoint dictates this. I believe the problem is that the multicopter still wants to stay in it's initial position and therefore commands pitch up. This problem was solved in position control mode which works fine in simulation.

tumbili, 2016-01-13:
@AndreasAntener @SimonWilks @LorenzMeier @sanderux FYI

sanderux, 2016-01-14:
I seem to have an issue leveling the quad. I had to put the frame several degrees off level to have it level in manual hover. I don't know what this will do to the FW mode. 

i also think trimming should be possible seperately for MC and FW. TRIM_ROLL/PITCH is available for FW but they must now be set manualy because copying trims stores for both MC and FW.

tubeme, 2016-01-21:
We've had similar experience for a month with the quad pitching up not only in mission but in manual flight transition until 1.1.0. It was in the master branch which we used to fly until 1.1.0. And the pitching of the quad got so intense fighting the plane, that we had a lot of 90 straight up stalls during the transition. We found a way to control it though...Just pulled the throttle down during the transition to make the quad spin on idle speed then when transition finished we pushed the throttle back in order to gain speed for the plane. I don't know if it is related or not... 

After 1.1.0 things got better. Just a couple of days ago we started experiencing back transition problems. The moment we switch to back transition the quad just wants to ditch the VTOL in the ground with almost full pitch down command. We double checked all the CGs and they are correct. Still have to make more flights to confirm consistency of the problem.

LorenzMeier, 2016-01-24:
This is to track the yaw issue:
https://github.com/PX4/Firmware/issues/3564

LorenzMeier, 2016-01-24:
This is to track the altitude issue:
https://github.com/PX4/Firmware/issues/3581

sanderux, 2016-01-30:
@tumbili @AndreasAntener 
I found the pitching up behavior that sitl has also happens in my airframe. In SITL setting FW_PSP_OFF to -25 gives it a somewhat straight transition. I also found strange and dangerous behavior when the takeoff wp and transition wp are inside the acceptance radius.

sanderux, 2016-01-30:
When reaching the waypoint that comes after a transition command, while the transition has not finished AND the waypoint after that is located on a different heading, the transition fails and the MC continues the mission, sometimes with oscillating pitch behavior.

As this seems inevitable i think the proper action would be to check this in the feasibility checker. Alternatively the mission handler could monitor the transition and fail/RTL if the next WP was reached before transition had completed.

AndreasAntener, 2016-02-02:
@tumbili Issue one on this list will probably be fixed by your proper TECS handling? https://github.com/PX4/Firmware/pull/3652

PR #3598 (2016-02-08): VTOL mission item handling
In general this will fix handling for 2 part mission items (takeoff, landing, transitions)
Fixes #3583
Fixes #3581
Fixes #3501
Fixes #3414

Should handle the following cases consistently:
 - [x] transition: yaw into direction of next wp then transition
 - [x] back transition: transition then move to wp
 - [x] back transition yaw issue
 - [x] takeoff: climb then move to wp
 - [x] land: move to wp then descend (fixes #3388)
 - [x] back transition: yaw north when flying south
closed 2016-02-08 by AndreasAntener