PX4/PX4-Autopilot #3505: Aiframe Config meta data must update to new naming and icon set

DonLakeFlyer, 2016-01-13:
This QGC pull https://github.com/mavlink/qgroundcontrol/pull/2602 switches to using the new icon set and naming as specified here: https://github.com/mavlink/qgroundcontrol/issues/2599#issuecomment-171424227.

QGC still has back compat for the old names.

DonLakeFlyer, 2016-01-13:
Another I noticed was the VTOL Duo and Quad where in the same group. That doesn't really work with the images as they stand now.

LorenzMeier, 2016-01-13:
File name:

```
+        <!-- Aiframe images supported by airfame meta data ml image tag -->
 +        <file alias="Airframe/AirframeSimulation">src/AutoPilotPlugins/Common/Images/AirframeSimulation.svg</file>
 +        <file alias="Airframe/AirframeUnknown">src/AutoPilotPlugins/Common/Images/AirframeUnknown.svg</file>
 +        <file alias="Airframe/Boat">src/AutoPilotPlugins/Common/Images/Boat.svg</file>
 +        <file alias="Airframe/FlyingWing">src/AutoPilotPlugins/Common/Images/FlyingWing.svg</file>
 +        <file alias="Airframe/Helicopter">src/AutoPilotPlugins/Common/Images/Helicopter.png</file>
 +        <file alias="Airframe/HelicopterCoaxial">src/AutoPilotPlugins/Common/Images/HelicopterCoaxial.png</file>
 +        <file alias="Airframe/HexaRotorPlus">src/AutoPilotPlugins/Common/Images/HexaRotorPlus.svg</file>
 +        <file alias="Airframe/HexaRotorX">src/AutoPilotPlugins/Common/Images/HexaRotorX.svg</file>
 +        <file alias="Airframe/OctoRotorPlus">src/AutoPilotPlugins/Common/Images/OctoRotorPlus.svg</file>
 +        <file alias="Airframe/OctoRotorPlusCoaxial">src/AutoPilotPlugins/Common/Images/OctoRotorPlusCoaxial.svg</file>
 +        <file alias="Airframe/OctoRotorX">src/AutoPilotPlugins/Common/Images/OctoRotorX.svg</file>
 +        <file alias="Airframe/OctoRotorXCoaxial">src/AutoPilotPlugins/Common/Images/OctoRotorXCoaxial.svg</file>
 +        <file alias="Airframe/Plane">src/AutoPilotPlugins/Common/Images/Plane.svg</file>
 +        <file alias="Airframe/PlaneATail">src/AutoPilotPlugins/Common/Images/PlaneATail.svg</file>
 +        <file alias="Airframe/QuadRotorATail">src/AutoPilotPlugins/Common/Images/QuadRotorATail.png</file>
 +        <file alias="Airframe/QuadRotorH">src/AutoPilotPlugins/Common/Images/QuadRotorH.png</file>
 +        <file alias="Airframe/QuadRotorPlus">src/AutoPilotPlugins/Common/Images/QuadRotorPlus.svg</file>
 +        <file alias="Airframe/QuadRotorVTail">src/AutoPilotPlugins/Common/Images/QuadRotorVTail.png</file>
 +        <file alias="Airframe/QuadRotorWide">src/AutoPilotPlugins/Common/Images/QuadRotorWide.svg</file>
 +        <file alias="Airframe/QuadRotorX">src/AutoPilotPlugins/Common/Images/QuadRotorX.svg</file>
 +        <file alias="Airframe/Rover">src/AutoPilotPlugins/Common/Images/Rover.svg</file>
 +        <file alias="Airframe/VTOLDuoRotorTailSitter">src/AutoPilotPlugins/Common/Images/VTOLDuoRotorTailSitter.svg</file>
 +        <file alias="Airframe/VTOLPlane">src/AutoPilotPlugins/Common/Images/VTOLPlane.svg</file>
 +        <file alias="Airframe/VTOLQuadRotorTailSitter">src/AutoPilotPlugins/Common/Images/VTOLQuadRotorTailSitter.svg</file>
 +        <file alias="Airframe/VTOLTiltRotor">src/AutoPilotPlugins/Common/Images/VTOLTiltRotor.svg</file>
 +        <file alias="Airframe/Y6A">src/AutoPilotPlugins/Common/Images/Y6A.png</file>
 +        <file alias="Airframe/Y6B">src/AutoPilotPlugins/Common/Images/Y6B.png</file>
 +        <file alias="Airframe/YMinus">src/AutoPilotPlugins/Common/Images/YMinus.svg</file>
 +        <file alias="Airframe/YPlus">src/AutoPilotPlugins/Common/Images/YPlus.svg</file>
```

DonLakeFlyer, 2016-01-13:
The image names in the meta data file don't include the "Airframe/" prefix.

LorenzMeier, 2016-01-13:
So they should or should not?

DonLakeFlyer, 2016-01-13:
Not
```
  <airframe_group image="HelicopterCoaxial" name="Coaxial Helicopter">
    <airframe id="15001" maintainer="Emmanuel Roussel" name="Coaxial Helicopter (such as Esky Lama v4 or Esky Big Lama)">
      <maintainer>Emmanuel Roussel</maintainer>
      <type>Coaxial Helicopter</type>
      <output name="MAIN1">Left swashplate servomotor, pitch axis</output>
      <output name="MAIN2">Right swashplate servomotor, roll axis</output>
      <output name="MAIN3">Upper rotor (CCW)</output>
      <output name="MAIN4">Lower rotor (CW)</output>
    </airframe>
  </airframe_group>
```

LorenzMeier, 2016-01-24:
Done.

closed 2016-01-24 by LorenzMeier