PX4/PX4-Autopilot #3509: Navio/ErleBrain 2 baseline Linux configuration

mhkabir, 2016-01-14:
Opening this to track our work on getting the POSIX port running on RPi2 hardware and either of those two shields.

LorenzMeier, 2016-01-14:
Thanks for digging through the docs!

LorenzMeier, 2016-01-14:
@ivereninov It would be fantastic to get a NAVIO 2 to Kabir. We have very much a focus on ports this month.

vmayoral, 2016-01-14:
Thanks @mhkabir for putting this together. A few additional bits about Erle-Brain 2 and PXFmini:

## Erle-Brain 2:
Target Board : RPi 2

### Sensors :
- MPU9250 IMU (SPI, `/dev/spidev0.1`)
- MS5611 Barometer (SPI, `/dev/spidev0.0`)
- PCA9685 PWM driver (I2C,` 0x40`)

### Buses :
- SPI : `/dev/spidev0.1`, `/dev/spidev0.0`
- I2C : `/dev/i2c-1`
- UART : `/dev/ttyAMA0`

### Docs and support:
- [Erle-Brain 2 docs](http://erlerobotics.com/docs/Artificial_Brains_and_Autopilots/Erle-Brain_2/Intro.html)
- [Erle Robotics support](http://forum.erlerobotics.com/)

## PXFmini:
Target Board : RPi0, RPi 2

### Sensors :
- MPU9250 IMU (SPI, `/dev/spidev0.1`)
- MS5611 Barometer (SPI, `/dev/spidev0.0`)
- PCA9685 PWM drivwqer (I2C,` 0x40`)

### Buses :
- SPI : `/dev/spidev0.1`, `/dev/spidev0.0`
- I2C : `/dev/i2c-1`
- UART : `/dev/ttyAMA0`

### Docs and support:
- [PXFmini docs](http://erlerobotics.com/docs/Artificial_Brains_and_Autopilots/Autopilot_shields/PXFmini/Intro.html)
- [Erle Robotics support](http://forum.erlerobotics.com/)

bo-rc, 2016-01-17:
Following! I want to contribute. Let me know what i can do. Thanks!

mhkabir, 2016-01-17:
Native build instructions are up here : https://pixhawk.org/modules/linux/rpi2
You would require my branch from here : <TODO>

Compilation is relatively fast, even natively, so this setup works out OK for now. The rpi2 build target is stripped down for now.

For cross-compiling, we need a 4.8 cross-toolchain, which I don't think the Pi Foundation provides. Help ? @vmayoral @Clouded @ivereninov

mhkabir, 2016-01-17:
Current state with native build : 
```
pi@navio-rpi:~/PX4/Firmware$ 
[init] shell id: 1995776000
[init] task name: mainapp

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

Ready to fly.

pxh> 
```

mhkabir, 2016-01-17:
@vmayoral @bo-rc @Clouded Everyone please see #3527

The cross-compiler and native build targets work fine now.

abencomo, 2016-01-17:
Hi Kabir,

Do you think we can give this a try on the RPi2 with the Navio+?

mhkabir, 2016-01-18:
@abencomo Yes you can :) Feel free to jump in! Just don't expect any hardware to work just yet.

bo-rc, 2016-01-22:
@mhkabir Are we going to take assignments for the drivers?

gelazanskas, 2016-03-04:
+1, waiting for PX4 to work with Navio 2

maxekman, 2016-04-18:
Following the with great interest!

Just wanted to mention that the link to the RT Linux by Emlid points to the older Navio (1) version, the Navio 2 version (for RPi 3) is here: http://docs.emlid.com/navio2/Navio-APM/configuring-raspberry-pi/

awright424, 2016-04-18:
I would also like to know when I can try this out my on rpi2/3

LorenzMeier, 2016-12-26:
Available on master.

closed 2016-12-26 by LorenzMeier