PX4/PX4-Autopilot #3511: 'module' object has no attribute 'Interpreter'

atelepin, 2016-01-14:
I try to build firmware and fail with next error:

 File "/home/soft/Autopilots/Firmware/Tools/genmsg/src/genmsg/template_tools.py", line 76, in _generate_from_spec
    interpreter = em.Interpreter(output=ofile, globals=g, options={em.RAW_OPT:True,em.BUFFERED_OPT:True})
AttributeError: 'module' object has no attribute 'Interpreter'
CMakeFiles/msg_gen.dir/build.make:301: recipe for target 'src/platforms/nuttx/px4_messages/px4_output_pwm.h' failed
make[3]: *** [src/platforms/nuttx/px4_messages/px4_output_pwm.h] Error 1

I try reinstall em module but with no success.
In source code of the em module i can't find any sign of Interpreter.

LorenzMeier, 2016-01-14:
What is the output of python --version? What is the OS and OS version you are using?

atelepin, 2016-01-15:
Python 2.7.11
Linux atelepin-nb3 4.2.0-23-generic #28-Ubuntu SMP Sun Dec 27 17:47:31 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
Kubuntu 15.10

The same issue i experience in kubuntu 12.04

atelepin, 2016-01-18:
SOLVED! In my distribution i have two version of "em", one was installed as egg package. I just delete it.

closed 2016-01-24 by LorenzMeier