PX4/PX4-Autopilot #3515: Off-board: sending local x,y,z from RPi

ArtemVass, 2016-01-15:
Hello,
I use RPi to compute local position indoors but I stuck on sending those to PixHawk as Mavlink packets. I use MAVLINK_MSG_ID_LOCAL_POSITION_NED 32 to transmit coordinates but I see no effect in MAVLinkInspector. By the way, setpoint messages are working fine.

sanderux, 2016-01-15:
did you enable MAV_DATA_STREAM_POSITION?

ArtemVass, 2016-01-15:
According to official docs:
THIS INTERFACE IS DEPRECATED AS OF JULY 2015. Please use MESSAGE_INTERVAL instead. A data stream is not a fixed set of messages, but rather a recommendation to the autopilot software. Individual autopilots may or may not obey the recommended messages.

Would be great if you could show the way to enable it!

sanderux, 2016-01-15:
Are you not getting data on POSITION_TARGET_LOCAL_NED?

ArtemVass, 2016-01-15:
Well, I do. I got data on both LOCAL_POSITION_NED(I get x,y,z where x,y are always 0 and z is a barometer measurement) and POSITION_TARGET_LOCAL_NED. So, I have no idea, what's wrong.

mhkabir, 2016-01-15:
Use VISION_POSITION_ESTIMATE or ATT_POS_MOCAP  to send your estimates.

closed 2016-02-02 by ArtemVass