PX4/PX4-Autopilot #3516: MC Takeoff

tumbili, 2016-01-15:
This is a list of corner cases / problems:

 - [ ] if max z speed is below takeoff jump speed then things go bad

ecmnet, 2016-01-15:
Comment from my side: It might be a good idea to separate the code of state control, setpoint calculation and target-attitude calculation a bit more. At least for me it was quite hard to follow the different modes of operation within mc_pos_control_main.cpp. This would also reduce the risk of side effects...

tumbili, 2016-01-15:
@ecmnet Generally I agree with you. However, if you want to do this you will have to test everything through again, which needs time.

ecmnet, 2016-01-15:
Yes, you're right. We will see, whether the list of issues grow. If it does and it is faster than resolving them, it would be time for refactoring.

edigotlieb, 2016-01-15:
happened to me too - caused a deadly crash because i set the takeoff altitude to 10m

LorenzMeier, 2016-01-17:
@tumbili Not sure we should add an acceptance radius in QGC for takeoff. Now that I'm thinking about it I even think the Firmware should always use its default. That will always yield a good value for all platform types and depend properly on the flight mode for VTOL. These are all things that QGC doesn't know about. So my proposal would be to ignore the acceptance radius completely, in all cases.

tumbili, 2016-02-14:
Probably everything fixed, need revaluation anyway.

closed 2016-02-14 by tumbili