PX4/PX4-Autopilot #3520: Mag calibration on current master is broken

SimonWilks, 2016-01-15:
Master version: 4ab39725ddbf728a81a8d5c3d2c1b0fee1be0cd2 
QGC: current master and recent daily

Starting mag calibration in QGC fails to start the calibration process.

SimonWilks, 2016-01-15:
Subsequent attempts after a restart seem to work, so it's possibly intermittent. Memory usage seems OK.

SimonWilks, 2016-01-15:
It's back. After setting up an airframe from scratch (could be coincidental) now it fails over several restarts. Seeing the following in the debug console which looks like something sensor related might be broken. Using beta it starts cleanly.
```
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
[i] microSD mounted: /fs/microsd
param: selected parameter default file /fs/mtd_params
[param] Loaded: /fs/mtd_params
rgbled on I2C bus 2 at 0x55 (bus: 100 KHz, max: 100 KHz)
 1:   SYS_USE_IO: match
 13005: + SYS_AUTOSTART: match
px4io: CRCs match
dataman: Power on restart, data manager file '/fs/microsd/dataman' size is 103090 bytes
MS5611_SPI on SPI bus 2 at 3 (11000 KHz)
ms5611: no device on bus 3
ms5611: driver start failed
bst: no devices found
adc init done
ver hwcmp match: PX4FMU_V2
hmc5883: no device on bus 2
hmc5883: no device on bus 1
mpu6000: driver start failed
MPU6000 on SPI bus 1 at 4 (1000 KHz)
L3GD20 on SPI bus 1 at 1 (11000 KHz)
LSM303D on SPI bus 1 at 2 (11000 KHz)
Airspeed on I2C bus 1 at 0x28 (bus: 100 KHz, max: 100 KHz)
sensors: ERROR: no barometer found on /dev/baro0 (2)
sensors: FATAL: No barometer found: /dev/baro0 (2)
sensors: sensor initialization failed
sensors: start failed
px4io default PWM output device
mavlink_if0: mode: 0, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
mavlink_if0: offboard mission init: ERROR
ver hwcmp match: PX4FMU_V2
sdlog2: log buffer size: 12288 bytes
[init] Vehicle type: VTOL
[i] Mixer: /etc/mixers/vtol_quad_x.main.mix on /dev/pwm_output0
[i] Mixer: /etc/mixers/vtol_AAERT.aux.mix on /dev/pwm_output1
WARN  started

ver hwcmp match: PX4FMU_V2
px4flow [187:100]
px4flow: scanning I2C buses for device..
mavlink_if1: mode: 5, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B

NuttShell (NSH)
nsh> WARN  Q POLL TIMEOUT
WARN  INAV poll timeout
WARN  Q POLL TIMEOUT
WARN  INAV poll timeout
commander_low_prio: settings saved.
```

SimonWilks, 2016-01-15:
@kd0aij could it be #3502 broke something here? Up till 32626b57a42f06fbd2d0a17532590c2b9906a2cc everything is ok. Haven't finished looking through it yet but there are some changes that look like they could play a role...

LorenzMeier, 2016-01-15:
Fixed here: https://github.com/PX4/Firmware/commit/415e42f5de5b52c411a260aa9911f28bee91c624

closed 2016-01-15 by LorenzMeier