PX4/PX4-Autopilot #3528: FMUv4 voltage scaling

LorenzMeier, 2016-01-17:
@kd0aij Do you have a power module? Is the scaling wrong?
https://github.com/PX4/Firmware/blob/master/src/modules/sensors/sensors.cpp#L855

kd0aij, 2016-01-18:
If we're talking about the new 8S ACSP4, I'm sure the scaling is different from the 3DR module.
I'll validate this when I get a chance to install an R12 and ACSP4 in a quad. I think Phil has already calculated and verified the scale factor.

LorenzMeier, 2016-01-18:
@pkocmoud Users report incorrect voltage readings. So I really need to know what the right scales are (and this is another thing that dropped, because I don't recall signing off on the final scaling values or having a chance of discussing it with the community).

kd0aij, 2016-01-19:
I'm seeing voltage_battery=0 in SYS_STATUS with ACSP4 connected using only one cable.
The BATTERY_STATUS message is all zero and -1, so it must not be getting updated.

LorenzMeier, 2016-01-19:
Normal customers seem to get voltage out. Can't comment without hardware.

kd0aij, 2016-01-19:
just measured voltage sense output of 742.7mV for input of 10.01V to ACSP4
ratio is .0742
full scale would be about 44.5V
8S is 8*4.2 = 33.6V max

But the firmware shows vehicle_status.voltage_battery = 0

LorenzMeier, 2016-01-23:
What is the scaling for current?

LorenzMeier, 2016-01-24:
I fixed it myself now in b2237ce.

closed 2016-01-24 by LorenzMeier