PX4/PX4-Autopilot #3534: yaw drift in multicopter attitude control

kd0aij, 2016-01-18:
I don't think any PID parameters have changed in this setup, but in recent flights it has not been holding heading very well
[image]

log: http://logs.uaventure.com/view/KAkQvSbUDpeabvQqYhEV7J

kd0aij, 2016-01-18:
Forgot to mention that I'm also seeing a large lag in yaw response. The copter continues to yaw after I release the stick, as though it is a poorly tracked heading setpoint which I'm controlling, instead of the actual yaw rate.

jgoppert, 2016-01-25:
This might be the same bug that is visible is SITL that causes the yaw to wander. It seems to be related to INAV. @kd0aij, can you try flying with LPE to confirm the bug is within INAV?

priseborough, 2016-01-26:
I observed the same issue when I flew today with EKF2

LorenzMeier, 2016-01-26:
@priseborough Can you share a log?

LorenzMeier, 2016-01-26:
On http://logs.uaventure.com and provide the link via Skype or here.

priseborough, 2016-01-26:
http://logs.uaventure.com/view/xM8RPz26NeFtmcR8Cnm5RL

LorenzMeier, 2016-01-26:
Logged in #3600.

closed 2016-01-26 by LorenzMeier