PX4/PX4-Autopilot #3535: Check mixer limitation

julianoes, 2016-01-18:
@muharred reported a possible bug in https://github.com/PX4/Firmware/blob/master/src/modules/systemlib/mixer/mixer_multirotor.cpp#L304

LorenzMeier, 2016-03-24:
I don't see what's wrong with that line - would be good to know what he thinks is broken there.

muharred, 2016-03-24:
I think the pointer was moved - should have used the commit-specific pointer. The problem is with the line bellow: https://github.com/PX4/Firmware/blob/791ec37f43635947dbe99e43b746144f30f1e442/src/modules/systemlib/mixer/mixer_multirotor.cpp#L310
max_out should be greater than 0.0 in all cases of positive thrust.

LorenzMeier, 2016-03-24:
Do you have a proposed fix?

muharred, 2016-03-24:
Well, I think `max_out > 1.0f` was intended and 0.0f got there as a copy-pase error.

LorenzMeier, 2016-03-24:
Yep. Fixed in 76a8bea.

closed 2016-03-24 by LorenzMeier