PX4/PX4-Autopilot #3542: Gazebo SITL POSCTL instability

julianoes, 2016-01-19:
The position control performance in Gazebo SITL seems to be impaired, at least on my machine. This leads to overshooting the waypoints (or manual commands) and "toilet bowling".

This needs further profiling and investigating.

[image]

tumbili, 2016-01-19:
@julianoes Good place to add that we need to check if the simulator puts a delay on the gps signal. Otherwise we would be compensating for a delay which is not there.

julianoes, 2016-01-19:
@tumbili Oh good go know. Can you point me to the place where the delay is compensated for?

LorenzMeier, 2016-01-25:
@julianoes @tumbili Can you convert all configs to EKF2 tomorrow and quickly test them? Including VTOL transitions.

tumbili, 2016-02-14:
Done, closing!

closed 2016-02-14 by tumbili