PX4/PX4-Autopilot #3543: Pixfalcon telemetry port rate can't be changed by SD card extras.txt file

sirPerna, 2016-01-19:
Telemetry rate can't be changed with this command inside extras.txt file:
mavlink start -d /dev/ttyS1 -b 57600 -r 3000

Also the parameter SYS_COMPANION = 0 doesn't stop the telemetry stream. This could be the source of the issue.

julianoes, 2016-01-19:
I believe you can only start additional mavlink instances but you can't change an existing one.

What are you trying to achieve, changing the baudrate or the rate? Maybe we could add an option to the mavlink cli to make it happen.

sirPerna, 2016-01-19:
I want to change the global rate and if possible the update rate for attitude and global position int frame. The default rate is too low for me.

mhkabir, 2016-01-19:
Why don't you simply set SYSTEM_COMPANION and use TELEM2 ?

sirPerna, 2016-01-19:
Pixfalcon has only ttyS1. Isn't SYSTEM_COMPANION mostly for baudrate and not for datarate?

mhkabir, 2016-01-19:
Ah I see. SYSTEM_COMPANION is only for TELEM2; It starts a high baud / high datarate onboard connection on S2.
In your case, a quick hack to do what you need would be adding a ```mavlink stop-all``` before ```mavlink start -d /dev/ttyS1 -b 57600 -r 3000```

sirPerna, 2016-01-19:
This works for me. Thank you!

closed 2016-06-21 by sirPerna

0xd1ma, 2017-03-09:
hello

I`m trying to change baudrate on Pixfalcon TELE1

On SD cart created: 
/etc/extras.txt

Inside extras.txt:
mavlink stop-all
mavlink start -d /dev/ttyS1 -b 921600 -r 20000

then trying to connect with mavros:
roslaunch mavros px4.launch fcu_url:="serial:///dev/ttyACM0:921600"

but it does`t work.

please help with it

julianoes, 2017-03-09:
@0xd1ma please open up a new issue instead of re-using this old thread.