PX4/PX4-Autopilot #3544: Pixracer Issues

tumbili, 2016-01-19:
- [ ] gps and mag cable are not wired correctly
 - [ ] need I2C splitter with power for e.g. airspeed sensor
 - [ ] rc calibration is lost sometimes (sombody needs to confirm this)
 - [ ] configuration error: no airspeed sensor

LorenzMeier, 2016-01-24:
@tumbili We need to work out what is going on with your airspeed sensor.

LorenzMeier, 2016-01-24:
@kd0aij Could you maybe create a splitter cable for your GPS and see if you can get GPS with Mag plus airspeed to work?

kd0aij, 2016-01-24:
next on my list :)

kd0aij, 2016-01-24:
I have 3DR GPS + mag, RGB LED and airspeed sensor working on R12, with all I2Cs in parallel on the GPS connector.
Booted several times USB only and power module only with no problems.
"tests sensors" reports "tests: command not found" though.

pkocmoud, 2016-01-24:
I can confirm using the current master (commit 6eac78d67529e20c3170092fd271c11c57c70f2f
)

tests sensors
nsh: tests: command not found

kd0aij, 2016-01-24:
the command name changed from "tests" to "test", but help still says "tests"

LorenzMeier, 2016-01-25:
The result also is printed if not enough memory is remaining (and the tests command has massive RAM requirements). We probably need to trim it down and that's all.

kd0aij, 2016-01-25:
I noticed that builitn_commands.c calls for 60,000 bytes of stack, and there certainly isn't that much free RAM.
Reduced the stack requirement by a factor of 10 to 6000 and "tests all" runs without hanging.
"tests sensors" output looks ok too. (in src/systemcmds/tests/CMakeLists.txt)

kd0aij, 2016-01-25:
Still have observed no failures of RGB LED or 3DR digital airspeed sensor.

kd0aij, 2016-01-25:
@tumbili it appears that something changed when I was trying to calibrate using QGC. Today I'm getting messages that no airspeed sensor is detected under any conditions that I've tried so far. Perhaps that is what the message "inspect sensors" was complaining about yesterday.
This might say something about the I2C sensor problem. The LED and airspeed sensor were detected reliably until I went through the sensor cal procedure for accel, gyro and compass. Could also be the choice of airframe. Pretty sure I used QGC to select a fixed-wing airframe when I was first testing the airspeed sensor.

kd0aij, 2016-01-25:
This shows the problem with I2C on the PixRacer. The airspeed sensor and RGB LED are powered with Vdd=5V, hence a valid logic high level is 3.5V. But the bus master is generating valid levels for Vdd=3.3V and SCL/SDA are never reaching a valid high level for the 5V devices. Reducing Vdd to 4.6V alleviates the problem by reducing VIH to .7*4.6=3.22V but there is very little noise margin in that case.

Logic thresholds obtained from I2C spec here (sec 3.1.2 and fig 38): http://www.nxp.com/documents/user_manual/UM10204.pdf
 
[image]

LorenzMeier, 2016-01-25:
@kd0aij If you raise your pull-up current (adding additional pull-ups), does it fix the issue? Your point regarding the reference manual makes perfect sense and its the kind of well-researched issue that really advances insight.

I would like to understand why this works on Pixhawk though (even if it might be marginal). So until I do, I'm not yet convinced we found the root cause.

kd0aij, 2016-01-26:
Testing with PH1 shows that the variance of diff_pressure (using QGC, units specified as "float") increases from about 20 with USB power to about 100 when powered by the 3DR power module. 
Could this indicate communication errors resulting from poor noise margin on the I2C bus, becoming worse at higher values of VDD for the MS4525DO sensor?
SCL and SDA levels look identical to those for the PixRacer, but there appears to be less undershoot on the falling edges.

[image]

pkocmoud, 2016-01-26:
[image]
Airspeed Sensor Vcc 3.3v

pkocmoud, 2016-01-26:
[image]
Airspeed Sensor Vcc 5.0v

pkocmoud, 2016-01-26:
Using meas_airspeed info, Read errors almost disappear with a Sensor VCC below 4.7v. At 5.0v I see 1000s of read errors per minute. Above 4.7v at the sensor and meas_airspeed start routinely fails to start due to sensor not being detected. I have ordered some i2c level shifters to try as an additional test.

pkocmoud, 2016-01-26:
[image]
Airspeed Sensor Vcc 5.0v with the 120R resistors added to the i2c data lines.

pkocmoud, 2016-01-26:
The intermittent detection of the sensor when attempting to start the "meas_airspeed start" still exists with the 120R resistors added at 5V. When it does start, read errors persist. 
At 3.3V Detection is 100% and read errors were 1 in 7000 reads.

pkocmoud, 2016-01-26:
[image]
Airspeed Sensor Vcc 3.3v with the 120R resistors added to the i2c data lines.

LorenzMeier, 2016-01-26:
So conclusion: Wrong design of the airspeed sensor breakout, its missing a PCA type I2C shifter from 3.3V to 5V. There might be more to it, but we can safely say that this is a critical contributor. Nothing looks wrong on the Pixracer schematic.

LorenzMeier, 2016-01-26:
I'm closing this since no Pixracer issue. I filed a new one:
https://github.com/PX4/Hardware/issues/55

@kd0aij @pkocmoud Thanks for the scope shots!

closed 2016-01-26 by LorenzMeier