PX4/PX4-Autopilot #3551: Sensor data coordinate frame

scott-eddy, 2016-01-20:
All, I'm having some issues parsing frames associated with MAVLINK data.  Specifically, the HIGHRES_IMU message.  The HIGHRES_IMU MAVLINK message parses the sensor_combined uORB message into the proper fields.  Looking at the .msg specification for sensor_combined it says that the [accel data](https://github.com/PX4/Firmware/blob/master/msg/sensor_combined.msg#L38), gyro data, and mag data are expressed in the body NED frame.  

However, the transformation applied to the data transforms it from the sensor frame to the body-fixed frame.  There is never an additional transformation from the body-fixed frame to the body NED frame.  Is the .msg comment wrong or is there another rotation applied elsewhere?

LorenzMeier, 2016-01-20:
All calculations are already done in body NED frame.

scott-eddy, 2016-01-20:
@LorenzMeier, I just want to be precise here.  The body-NED frame has origin at C.G. of aircraft, X-axis aligned with North, Y-axis aligned with East, Z-axis aligned with down correct?  I see the data as expressed in the body-fixed frame with origin at C.G., X-axis pointing out the nose of the aircraft, Y-axis aligned with right wing of aircraft, Z-axis down.  

I don't see a rotation from the body-fixed frame to the body-NED frame performed on the sensor data.  I see the rotation from the sensor board frame to the body-fixed frame but never from body-fixed to body-NED.  Wouldn't the second rotation require knowing the vehicle's current attitude?

kd0aij, 2016-01-20:
Isn't body-NED self contradictory? body frame and NED frame have different
axis orientations

scott-eddy, 2016-01-20:
I know it's old documentation but according to the [old wiki](https://pixhawk.org/dev/know-how/frames_of_reference) there are two body frames.  The body-fixed frame which is what most people think of when referring to the body frame (e.g. x out nose of air-frame, y to the right, z down).  The body-NED frame, according to the docs, is rigidly attached to the aircraft but has axis always aligned with the inertial NED frame regardless of aircraft orientation.  

I'm 99% sure that the body-NED frame is rarely, if ever, used in the firmware.  And I'm assuming that the msg specification meant the body-fixed frame but use "NED" to denote that Z was down.  We should update the msg file if I'm right.

kd0aij, 2016-01-20:
"body-fixed NED" vs. "earth-fixed NED" makes sense, although it's not clear
from that whether earth-fixed also implies earth-centered or some arbitrary
origin

scott-eddy, 2016-01-22:
@LorenzMeier @kd0aij The accelerations reported by the sensor_combined message are W.R.T a body-fixed frame with X-forward, Y-left, Z-up.  This is easily seen when looking at the Raw accelerations recorded by sdlog2 and noting that the Z acceleration is reported as -9.8 [m/s/s] implying a Z-UP frame.  **This body-fixed frame is not documented anywhere.**

However, the gyro readings are W.R.T a frame with X-forward, Y-right, Z-Down.  Documentation should be updated.

tumbili, 2016-01-22:
@scott-eddy https://en.wikipedia.org/wiki/G-force

kd0aij, 2016-01-22:
@scott-eddy I think Roman's point is that negative z accel is consistent with Z down, since the vehicle is accelerating UPWARD when at rest in the earth frame. Also the accel frame is most certainly the same as the gyro frame (and right-handed).

tumbili, 2016-01-22:
@scott-eddy I agree that the body NED frame is rarely used in the firmware. I think the difference between the body NED frame and the earth NED frame (apart from the translation) becomes only significant if you travel very large distances. But I agree, if we define these frames we should be very precise with the naming in the messages.

tumbili, 2016-01-22:
http://ch.mathworks.com/help/aeroblks/about-aerospace-coordinate-systems.html

scott-eddy, 2016-01-22:
@kd0aij @tumbili I see now that the frames are aligned with body-fixed Z down.  I don't understand why when g is aligned with a body axis you measure -9.8 m/s.  For instance when g is aligned with Z-down aren't you accelerating at +9.8m/s in the Down direction?  Some clarity would be greatly appreciated here, and I'll close the issue.

tumbili, 2016-01-22:
@scott-eddy If you imagine the accelerometer as a mass held in the air by two springs on the side then the effect of gravity would lead to the mass being pulled down. This is the same effect that you would get if you accelerate the sensor in an upwards direction (due to it's inertia the mass will be pulled DOWN). This is why the sensor shows a negative value because it feels an acceleration going UP but it used a NED frame to positive value is DOWN.

scott-eddy, 2016-01-22:
@tumbili Thanks.  Closing this

closed 2016-01-22 by scott-eddy

sardarrerhan45, 2022-09-20:
Hi all, Can anyone explain me how to rotate from sensor to body frame for vehicle acceleration , and what is sensor frame specifically? 
Thanks in advance!