PX4/PX4-Autopilot #3552: Error Building SITL Gazebo Simulation

tasarmiento, 2016-01-20:
Command:  make posix_sitl_default gazebo
Using Xcode 7.2

The following occurs:

git submodule init
Tools/check_submodules.sh 
Git submodule config valid.
Checked NuttX submodule, correct version found
Checked Tools/gencpp submodule, correct version found
Checked Tools/genmsg submodule, correct version found
Checked Tools/jMAVSim submodule, correct version found
Checked Tools/sitl_gazebo submodule, correct version found
Checked cmake/cmake_hexagon submodule, correct version found
Checked mavlink/include/mavlink/v1.0 submodule, correct version found
Checked src/lib/DriverFramework submodule, correct version found
Checked src/lib/dspal submodule, correct version found
Checked src/lib/ecl submodule, correct version found
Checked src/lib/matrix submodule, correct version found
Checked src/modules/uavcan/libuavcan submodule, correct version found
Checked unittests/googletest submodule, correct version found
ninja -C /Users/TAS/src/Firmware/build_posix_sitl_default  gazebo
ninja: Entering directory `/Users/TAS/src/Firmware/build_posix_sitl_default'
[15/27] Building CXX object src/systemcmds/topic_listener/CMakeFiles/systemcmds__topic_listener.dir/topic_listener.cpp.o
FAILED: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++   -DCLOCK_MONOTONIC=1 -DCONFIG_ARCH_BOARD_SITL -DPX4_MAIN=listener_app_main -D__DF_DARWIN -D__PX4_DARWIN -D__PX4_POSIX -Dnoreturn_function="__attribute__((noreturn))" -I../src -I. -Isrc -I../src/modules -I../src/include -I../src/lib -I../src/platforms -I../src/drivers/boards/sitl -Isrc/modules/px4_messages -Isrc/modules -I../mavlink/include/mavlink -I../src/lib/DriverFramework/framework/include -I../src/lib/matrix -I../src/modules/systemlib -I../src/lib/eigen -I../src/platforms/posix/include -Iexternal/Install/include -g -fno-exceptions -fno-rtti -std=gnu++0x -fno-threadsafe-statics -DCONFIG_WCHAR_BUILTIN -D__CUSTOM_FILE_IO__ -Wall -Werror -Wno-sign-compare -Wextra -Wfloat-equal -Wpointer-arith -Wmissing-declarations -Wno-unused-parameter -Werror=format-security -Werror=array-bounds -Wfatal-errors -Werror=unused-variable -Werror=reorder -Werror=uninitialized -Werror=init-self -Wno-unused-const-variable -Wno-varargs -Wno-missing-field-initializers -Os -fno-strict-aliasing -fomit-frame-pointer -funsafe-math-optimizations -ffunction-sections -fdata-sections -fvisibility=hidden -include visibility.h   -include visibility.h -Os -MMD -MT src/systemcmds/topic_listener/CMakeFiles/systemcmds__topic_listener.dir/topic_listener.cpp.o -MF src/systemcmds/topic_listener/CMakeFiles/systemcmds__topic_listener.dir/topic_listener.cpp.o.d -o src/systemcmds/topic_listener/CMakeFiles/systemcmds__topic_listener.dir/topic_listener.cpp.o -c src/systemcmds/topic_listener/topic_listener.cpp
src/systemcmds/topic_listener/topic_listener.cpp:164:11: fatal error: comparison of integers of different signs: 'int' and 'unsigned int' [-Wsign-compare]
                while(i < num_msgs) {
                      ~ ^ ~~~~~~~~
1 error generated.
[15/27] Building CXX object src/modules/mavlink/CMakeFiles/modules__mavlink.dir/mavlink_main.cpp.o
ninja: build stopped: subcommand failed.
make: *** [posix_sitl_default] Error 1

LorenzMeier, 2016-01-20:
Run make clean on latest master. This is already resolved but the file might not get regenerated properly.

LorenzMeier, 2016-01-20:
Fixed here: https://github.com/PX4/Firmware/commit/a49dbbc9a8207999b252ddca5210d35348f83afa

closed 2016-01-20 by LorenzMeier