PX4/PX4-Autopilot #3560: MC: Landing ground detection needs improvements

AndreasAntener, 2016-01-21:
2 corner cases
- sometimes the velocity change when hitting the ground triggers the free fall logic, maybe we need to be more allowing there (effect, it bounces around on the ground, see log)
- ground detection with a relatively slow landing speed of 0.5m/s can trigger in air. the standard vtol is heavily affected in wind because it gets additional lift and seems to get stuck at the same altitude sometimes on slow landings

http://logs.uaventure.com/view/9HR7JtSCDwyikcCybanS2V#Thrust_PLOT

AndreasAntener, 2016-02-19:
The behaviour was improved on master recently but the issues still exist.

dagar, 2017-04-12:
@Stifael @AndreasAntener any change here?

Stifael, 2017-04-13:
https://github.com/PX4/Firmware/issues/6730

Stifael, 2017-04-19:
I will close it since it is discussed here #6730

closed 2017-04-19 by Stifael