PX4/PX4-Autopilot #3564: MC, VTOL: corner cases in yaw handling during setpoint changes (mission)

AndreasAntener, 2016-01-22:
We've seen the following cases:
- takeoff: when switching from climbing to moving to the waypoint, it yaws into a (random) direction first
- transition: before the transition starts, it sometimes waits for a bit and yaws forth and back
- transition: at the moment of commanding the transition the yaw setpoint spikes

[image]
[image]

http://logs.uaventure.com/view/pry8kgGojZtXxnqGQPx2SU

LorenzMeier, 2016-01-24:
I pushed two fixes:
https://github.com/PX4/Firmware/commit/737fe1fc7f689456722a56cbd21bbb1844771931
and
https://github.com/PX4/Firmware/commit/081da8bb7f37c0a6a7323d85e19c0fd5b4283eeb

These were both definitely needed, as they represent architecture corner cases for VTOL. I'm not sure this fixes everything, but it would be good if @tumbili Could test this tomorrow morning in SITL.

LorenzMeier, 2016-01-24:
One more log:
http://logs.uaventure.com/view/8aaJzVRXvEnGfvLP63L5Pd

LorenzMeier, 2016-01-24:
You might need to fix this as well or first to unblock this:
https://github.com/PX4/Firmware/pull/3582

I suggest to set up a call with Andreas first.

CornerOfSkyline, 2016-01-24:
I found that the reset_yawsetpoint was not actived during the transition, the mc attitude control will try to control the yaw.

https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L1819

Though here set the reset_yaw_sp = true, but it did not work in the control loop.We need to reset the yaw setpoint before to publish the attitude sp

AndreasAntener, 2016-02-08:
Fixed via #3598

closed 2016-02-08 by AndreasAntener