PX4/PX4-Autopilot #3566: Fix acceptance radius for special waypoints (landing, takeoff)

AndreasAntener, 2016-01-22:
Discussed in #3516, there are certain situations where only a system defined acceptance radius makes sense. These need to be set consistently in mission and other auto modes.

LorenzMeier, 2016-01-24:
@AndreasAntener We should for takeoff and landing always use the default system radius.

AndreasAntener, 2016-01-24:
@LorenzMeier yes, I was thinking the system default should act as maximum, if NAV_ACC_RAD is smaller we take the smaller one. For a small fixed-wing we need to be able to make it smaller because the takeoff WP could be well within 50m (default) of starting point.

(but ignoring what is set on the waypoint)

LorenzMeier, 2016-01-24:
@AndreasAntener No, that's exactly the wrong way to think about it. It means you overshoot and the L1 controller can't do its job. The right fix is to only use the smaller one for rotary wing mode.

AndreasAntener, 2016-01-24:
I see your point, but if you have a small plane, you will also have a reduced L1 period. If it overshoots its not setup properly.

LorenzMeier, 2016-01-24:
Fixed in 06b496e. We might need to publish navigation_capabilities from the VTOL controller and consume the fixed wing and multi rotor capabilities in there.

I would not be surprised if that fixes your issues - where are we with master?

LorenzMeier, 2016-01-24:
Yes, but if your L1 period is smaller then the navigation capabilities number will be smaller as well. You can't do L1 navigation without locking the switch distance to the actual L1 tracking distance. You're violating the fundamentals of the algorithm then.

AndreasAntener, 2016-01-24:
I wasn't aware of the nav caps turn distance, makes sense. Basically:
- fixed-wing: use turn distance
- multicopter: use 2m (default)

This issue is also about cleaning up the inconsistencies in the use of `get_acceptance_radius` which at least for takeoff in MC it's currently wrong. I synced with @tumbili and I'll go through this and the yawing issues tomorrow.

LorenzMeier, 2016-01-24:
I'm not so sure if its really wrong. Did you really test with master? Code review looks good. In general the branch handling needs to improve a lot since there are way too many dangling branches and fixes.

Here is a PR you might want or need: https://github.com/PX4/Firmware/pull/3579

AndreasAntener, 2016-01-25:
Here's what I mean: https://github.com/PX4/Firmware/blob/master/src/modules/navigator/mission_block.cpp#L151-L175
For takeoff and landing we directly use the nav_acc_rad parameter, which might be wrong.

LorenzMeier, 2016-01-25:
That's fine. But it has to be set to multicopter defaults. So 2-3m

AndreasAntener, 2016-02-08:
Fixed via #3598

closed 2016-02-08 by AndreasAntener