PX4/PX4-Autopilot #3569: Revisit the VFR_HUD message content

sanderux, 2016-01-22:
In the 20160111 dev call an issue to track requested additions / changes in the VFR_HUD mavlink message was discussed. (https://pixhawk.ethz.ch/mavlink/#VFR_HUD)

Among others the following were noted;
- Altitude should be available as local, amsl and relative
- Battery voltage 
- Flight mode
- Distance to home
- Current mission WP nr
- Distance to next WP

sanderux, 2016-01-22:
@tumbili do you know / can you mention the members of that call that had idea's on this?

LorenzMeier, 2016-01-24:
We can't change messages without breaking the API. Closing this for now. The right forum for new MAVLink messages would be the MAVLink Google group or http://discuss.dronecode.org.

closed 2016-01-24 by LorenzMeier