PX4/PX4-Autopilot #3570: Failsafe procedure on sdcard error

sanderux, 2016-01-23:
The following mavlink log shows a vehicle that was operating with what appears te be a corrupt sdcard. 
[disaster.mavlink.zip](https://github.com/PX4/Firmware/files/101813/disaster.mavlink.zip)

The warnings are shown in this image:
[image]

It seems the vehicle did notice there was no valid mission and decided to loiter. The attempt to hold position failed. The mavlink log, seconds after entering mission mode,  shows the vehice as tilted left forward while the gps data shows the vehicle was moving as if it were tilted in the opposite direction.

[image]

LorenzMeier, 2016-01-24:
Parameters are not stored on the microSD card. They are entirely isolated from anything happening there.

sanderux, 2016-01-24:
@LorenzMeier Do you have any idea on how the sensor could become reversed? I'm trying to figure out if it is related to the dataman and corrupt SD card.

sanderux, 2016-01-24:
This is the flight path at the moment it should have been loitering
[image]

sanderux, 2016-01-24:
And this is the sensor reading at that moment
[image]

LorenzMeier, 2016-01-24:
It started to log successfully on the microSD card. I suspect therefore potentially a power related or mechanical issue. Unfortunately that's not transmitted at high enough rate via telemetry. Note the failure of the first magnetometer. If the frame was badly calibrated the failover to the 2nd mag might not have been enough to recover.

sanderux, 2016-01-24:
That would still not explain the reversal. This reversal appears to have happened at the moment it switched to mission or the pilot would never have been able to fly manual. Could that in any way be related to the corrupt sd?

LorenzMeier, 2016-01-24:
It switched to the internal mag at that point. If its environment was changed (something magnetic added in its vincinity) after calibration it could reverse the heading.

sanderux, 2016-01-24:
In that case there is nothing the firmware could have done to prevent this.
I would suggest a thorough analysis if a similar event is reported in the future.

I think we can close this one for now.

closed 2016-01-24 by sanderux