PX4/PX4-Autopilot #3575: Review RTL performance

LorenzMeier, 2016-01-24:
@julianoes @tumbili In this log the user reported that he had to abort RTL. I'm not sure yet what was exactly going on.

LorenzMeier, 2016-01-24:
RTL failure:
http://logs.uaventure.com/view/u4aZuZJYjeZuS3PwdGYBCF

Mission successful:
http://logs.uaventure.com/view/Rj4TSTuEZU3Y2mvwPeRjaJ

LorenzMeier, 2016-01-24:
@tumbili If we are missing data here to do the analysis properly (e.g. the home position) we should pimp our tools.

AndreasAntener, 2016-01-24:
We probably hit the same issue, triggering RTL via command after offboard flight. The climb to RTL height worked as expected but returning to home didn't behave.
http://logs.uaventure.com/view/4Am2auDFHwDXtCmc8CPEzR

LorenzMeier, 2016-01-24:
@julianoes @tumbili This is our absolute first priority. We need to nail this ASAP.

LorenzMeier, 2016-01-25:
@julianoes This is another log (linked above already): http://logs.uaventure.com/view/u4aZuZJYjeZuS3PwdGYBCF

LorenzMeier, 2016-01-25:
@julianoes @tumbili Since it affects mostly RTL I'm wondering if its related to home position yaw handling.

LorenzMeier, 2016-01-25:
@kd0aij Could you share a log of your yaw tracking issue here as well?

kd0aij, 2016-01-26:
https://github.com/PX4/Firmware/issues/3534#issue-127252193

AndreasAntener, 2016-01-27:
Fixed on master via c9b1fb154d3e6ea1bbb28401688811a59c0f6543

closed 2016-01-27 by AndreasAntener