PX4/PX4-Autopilot #3580: ros_sitl_default build broken?

julianoes, 2016-01-24:
I tried compiling ```make ros_sitl_default``` and ran into:

```
[ 96%] Building CXX object CMakeFiles/position_estimator.dir/src/platforms/ros/nodes/position_estimator/position_estimator.cpp.o
Building CXX object CMakeFiles/publisher.dir/src/examples/publisher/publisher_main.cpp.o
/home/julianoes/src/Firmware/src/platforms/ros/nodes/mc_mixer/mc_mixer.cpp:43:40: fatal error: mav_msgs/CommandMotorSpeed.h: No such file or directory
compilation terminated.
CMakeFiles/mc_mixer.dir/build.make:54: recipe for target 'CMakeFiles/mc_mixer.dir/src/platforms/ros/nodes/mc_mixer/mc_mixer.cpp.o' failed
make[3]: *** [CMakeFiles/mc_mixer.dir/src/platforms/ros/nodes/mc_mixer/mc_mixer.cpp.o] Error 1
CMakeFiles/Makefile2:1858: recipe for target 'CMakeFiles/mc_mixer.dir/all' failed
make[2]: *** [CMakeFiles/mc_mixer.dir/all] Error 2
```

Is this build target, platform still supported (and advised to be used)?

LorenzMeier, 2016-01-24:
No, its not showing up in our docs anywhere. You should be using ```posix_sitl_default gazebo```. I don't want to remove the target right away, but won't fix this.

closed 2016-01-24 by LorenzMeier

julianoes, 2016-01-24:
Well, people still find it somehow :(, ok.

LorenzMeier, 2016-01-24:
This should help with that. Thanks for raising this usability issue!
9169a58