PX4/PX4-Autopilot #3583: Navigator: Move action handling into read routine, never load actions as mission items

LorenzMeier, 2016-01-24:
@tumbili @AndreasAntener @julianoes Summarising my findings here again:

We check for action items here and publish the command:
https://github.com/PX4/Firmware/blob/0102e47708ff44d478cde919638b2f65b91ac0a6/src/modules/navigator/mission_block.cpp#L113-L127

The problem is that when we get there, the payload data from these items has already been taken and published as 3D coordinate. These are the bogus altitude values you see in VTOL.

The command publication code should be moved into this routine instead:
https://github.com/PX4/Firmware/blob/6eac78d67529e20c3170092fd271c11c57c70f2f/src/modules/navigator/mission.cpp#L658-L759

In the routine the code should cycle through the mission, publish all commands at it sees them (delaying between commands for 1000 us so they executed) and then return with the following 3D position it finds.

This way we will not have the twitch with VTOL any more. If you do this properly, these two PRs can be closed (as they try to fix it on the wrong location):
https://github.com/PX4/Firmware/pull/3579
this one should be harmless, but its a huge diff, so remove it rather as well:
https://github.com/PX4/Firmware/pull/3582

Let me know if this is not crystal clear and you need architectural help.

LorenzMeier, 2016-01-24:
@sanderux FYI

sanderux, 2016-01-26:
Could we treat action items as an integral part of the next mission item? if so then we could attach the action item to the mission item. The mission item could initialize all setpoints and after that activate the action item. This would remove the necessity to read ahead on mission items.

sanderux, 2016-01-26:
well, something like that yes..

AndreasAntener, 2016-01-26:
@LorenzMeier The branch I just added is WIP. I went over this and to me the `read_mission_item` routine doesn't seem to be the right place to put the command handling code. For the forward transition we want to read ahead anyway (so we can yaw into the right direction first), so I'm skipping over the command items one level above the read routine. The command handling happens where the other mission items get handled because it's the place where we know what to do with the next item. The setpoints are now protected against the non-position items. Also, because I think we want to wait on the status change eventually, handling the wait in `is_mission_item_reached` seems to be ok (added a temporary wait for now as you did in one of your quick fixes).

@sanderux Interesting thought, could be tricky with multiple commands between 2 waypoints

I also plan to handle the following cases consistently:
- transition: yaw into direction of next wp then transition
- back transition: transition then move to wp
- takeoff: climb then move to wp
- land: move to wp then descend

AndreasAntener, 2016-01-26:
Discussed with @sanderux. Will push more code asap so we get the bigger picture of the changes involved and go from there.

PR #3598 (2016-02-08): VTOL mission item handling
In general this will fix handling for 2 part mission items (takeoff, landing, transitions)
Fixes #3583
Fixes #3581
Fixes #3501
Fixes #3414

Should handle the following cases consistently:
 - [x] transition: yaw into direction of next wp then transition
 - [x] back transition: transition then move to wp
 - [x] back transition yaw issue
 - [x] takeoff: climb then move to wp
 - [x] land: move to wp then descend (fixes #3388)
 - [x] back transition: yaw north when flying south
closed 2016-02-08 by AndreasAntener