PX4/PX4-Autopilot #3585: Navigator Times out while sitting unarmed on the bench.

pkocmoud, 2016-01-24:
Using Master (https://github.com/PX4/Firmware/commit/6eac78d67529e20c3170092fd271c11c57c70f2f) and a PixRacer, After sitting idle on the bench for 20 minutes. Radio was off and Battery was disconnected. I heard a DatDat-Dat-Duh tone and the NSH console started to repeat: "WARN  navigator timed out". Is this a problem?

Here is my boot up until and including the previous message.

sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
[i] microSD mounted: /fs/microsd 
param: selected parameter default file /fs/mtd_params
[param] Loaded: /fs/mtd_params 
rgbled: no RGB led on bus #1
blinkm: I2C init failed
blinkm: init failed
 1:   SYS_USE_IO: match
ver hwcmp match: PX4FMU_V4
 10020: + SYS_AUTOSTART: match
ver hwcmp match: PX4FMU_V4
dataman: Unknown restart, data manager file '/fs/microsd/dataman' size is 103090 bytes
MS5611_SPI on SPI bus 2 at 3 (20000 KHz)
bst: no devices found
adc init done
ver hwcmp match: PX4FMU_V4
HMC5883_I2C on I2C bus 1 at 0x1e (bus: 100 KHz, max: 400 KHz)
HMC5883_SPI on SPI bus 1 at 5 (11000 KHz)
MPU6000 on SPI bus 1 at 6 (1000 KHz)
MPU9250 on SPI bus 1 at 4 (1000 KHz)
meas_airspeed: no MS4525 airspeed sensor connected
ets_airspeed: no ETS airspeed sensor connected
ets_airspeed: no ETS airspeed sensor connected
sensors: mag cal status changed 
sensors: mag cal status changed 
WARN  No RGB LED found at /dev/rgbled0
[i] FMU mode_pwm started 
ver hwcmp match: PX4FMU_V4
mavlink_if0: mode: 0, data rate: 800000 B/s on /dev/ttyS0 @ 921600B
mavlink_if0: hardware flow control not supported
mavlink_if0: offboard mission init: ERROR
mavlink_if1: mode: 0, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
ver hwcmp match: PX4FMU_V4
mavlink_if2: mode: 3, data rate: 1200 B/s on /dev/ttyS2 @ 57600B
sdlog2: log buffer size: 12288 bytes
[i] MULTICOPTER 
[i] Mixer: /etc/mixers/quad_x.main.mix on /dev/pwm_output0 
 1:   INAV_ENABLED: match
ver hwcmp match: PX4FMU_V4
frsky_telemetry: opening uart
mavlink_if3: mode: 5, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B

NuttShell (NSH)
nsh> gps: module found: UBX
mavlink_if1: Disabling hardware flow control
position_estimator_inav: init ref: lat=41.8539501, lon=-88.4545830, alt=244.6590
telem> home: 41.8542914, -88.4541761, 242.50
WARN  navigator timed out
WARN  navigator timed out
WARN  navigator timed out
WARN  navigator timed out
WARN  navigator timed out
WARN  navigator timed out
WARN  navigator timed out
WARN  navigator timed out
WARN  navigator timed out
WARN  navigator timed out
WARN  navigator timed out

LorenzMeier, 2016-01-24:
It sounds like something non-good happened on GPS lock. @kd0aij Could you verify that as fast as possible? Could be FrSky telemetry or GPS related.

@pkocmoud Could do this again, but this time also run "top" when it happens (or do you still have the board in that state?)

LorenzMeier, 2016-01-24:
@julianoes @tumbili This is something we need to settle ASAP as well. Looks like we start to move faster than we should.

I suspect either FrSky telemetry or this change:
https://github.com/PX4/Firmware/commit/a284b77c7b72b2f291878990f0e5ad17750273ab

@kd0aij If you can repro, try undoing that commit.

pkocmoud, 2016-01-25:
The timeout was much faster this time and no tone from the speaker.

nsh> position_estimator_inav: init ref: lat=41.8550376, lon=-88.4545191, alt=336.2960
WARN  navigator timed out
WARN  navigator timed out
WARN  navigator timed out
WARN  navigator timed out
WARN  navigator timed out
WARN  navigator timed out
Processes: 25 total, 3 running, 22 sleeping
CPU usage: 33.69% tasks, 0.98% sched, 65.32% idle
Uptime: 261.220s total, 170.603s idle

 PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE 
Processes: 25 total, 2 running, 23 sleeping
CPU usage: 37.77% tasks, 0.59% sched, 61.65% idle
Uptime: 262.582s total, 171.443s idle

Processes: 25 total, 3 running, 22 sleeping
CPU usage: 37.76% tasks, 0.52% sched, 61.73% idle
Uptime: 263.939s total, 172.280s idle

Processes: 25 total, 2 running, 23 sleeping
CPU usage: 38.20% tasks, 0.59% sched, 61.21% idle
Uptime: 265.296s total, 173.111s idle
Processes: 25 total, 5 running, 20 sleeping
CPU usage: 40.01% tasks, 0.66% sched, 59.32% idle
Uptime: 266.654s total, 173.916s idle
Processes: 25 total, 4 running, 21 sleeping
CPU usage: 39.79% tasks, 0.81% sched, 59.40% idle
Uptime: 268.012s total, 174.722s idle

 PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE 
   0 Idle Task                  174722 59.395     0/    0   0 (  0)  READY 
   1 hpwork                      12387  5.011   836/ 1592 192 (192)  w:sem 
   2 lpwork                        983  0.294   356/ 1592  50 ( 50)  READY 
   3 init                         1627  0.000  1284/ 2496 100 (100)  w:sem 
 199 top                           251  3.463  1212/ 1696 100 (100)  RUN   
  85 dataman                         1  0.000   676/ 1192  90 ( 90)  w:sem 
 104 sensors                     10103  3.684  1756/ 1992 250 (250)  w:sem 
 106 gps                           409  0.147   660/ 1192 220 (220)  w:sem 
 108 commander                    1215  0.368  3132/ 3592 140 (140)  w:sig 
 109 commander_low_prio              9  0.000   652/ 2872  50 ( 50)  w:sem 
 115 mavlink_if0                 11286  4.274  2372/ 2696 100 (100)  w:sig 
 116 mavlink_rcv_if0                21  0.000   812/ 2096 175 (175)  w:sem 
 119 mavlink_if1                  2359  0.810  2372/ 2696 100 (100)  w:sig 
 120 mavlink_rcv_if1                20  0.000   812/ 2096 175 (175)  w:sem 
 123 mavlink_if2                  2107  0.736  2364/ 2696 100 (100)  w:sig 
 124 mavlink_rcv_if2                21  0.000   812/ 2096 175 (175)  w:sem 
 134 sdlog2                        851  0.294  2060/ 3296  70 ( 70)  w:sig 
 153 attitude_estimator_q        12784  4.863  2012/ 2392 250 (250)  w:sem 
 157 position_estimator_inav      6297  3.389  4756/ 5296 250 (250)  w:sem 
 159 mc_att_control               6475  2.579   828/ 1496 250 (250)  w:sem 
 165 mc_pos_control               1667  0.663  1060/ 1496 250 (250)  w:sem 
 171 navigator                     251  0.884   804/ 1496 105 (105)  w:sem 
 178 frsky_telemetry              2582  1.031  1316/ 1992 200 (200)  w:sem 
 190 mavlink_if3                 19092  7.295  2372/ 2696 100 (100)  w:sig 
 196 mavlink_rcv_if3                39  0.000  1116/ 2096 175 (175)  w:sem

pkocmoud, 2016-01-25:
Not sure it is related, but after quitting top, the WARN is not occuring.

nsh> navigator status
navigator: Geofence is valid
nsh> telem> home: 41.8542642, -88.4543835, 304.36

kd0aij, 2016-01-25:
I saw navigator timeouts once, immediately after booting the config with 3DR GPS/compass, LED and airspeed sensor. Have not seen it occur again through several subsequent reboots issued via console and by cycling USB power.  GPS lock has not triggered it, and frsky_telemetry is running with an X4R RX connected. Also has not occurred on cold or warm boot on "brick" power.

LorenzMeier, 2016-01-25:
@pkocmoud I've now reverted the GPS change - let me know if the issue resolves if you flash through master.

If not, please inspect your board. We've now seen so many differences between boards that I would not be surprised if a power circuit / manufacturing difference could be at the bottom of this. One of my boards only outputs 1.8V on the GPS port. Can you measure the voltage your GPS gets?

davids5, 2016-01-25:
@Lorenz

 

Just to add some more data points.

 

In the last week I saw this on master with FMUV2 hw with no peripherals connected (only pwm).

 

I also saw 80% plus cpu utilization until I did a "mtd erase"

 

ekf2 was 36% and I think commander was the other task with 30%+.

LorenzMeier, 2016-01-26:
@pkocmoud Does this still reproduce for you with current master?

pkocmoud, 2016-01-27:
Sorry Lorenz, I have not had time to test yet.

pkocmoud, 2016-01-28:
Confirmed issue exists in Stable. Testing Master (cc7929937505c34d06b070c708e3f56a755cbda7) now.

pkocmoud, 2016-01-28:
It occurs in the before listed master.

Here is the Boot and Top if it helps:

�sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
[i] microSD mounted: /fs/microsd 
param: selected parameter default file /fs/mtd_params
[param] Loaded: /fs/mtd_params 
rgbled: no RGB led on bus #1
blinkm: I2C init failed
blinkm: init failed
 1:   SYS_USE_IO: match
ver hwcmp match: PX4FMU_V4
 10020: + SYS_AUTOSTART: match
ver hwcmp match: PX4FMU_V4
dataman: Unknown restart, data manager file '/fs/microsd/dataman' size is 103090 bytes
MS5611_SPI on SPI bus 2 at 3 (11000 KHz)
bst: no devices found
adc init done
ver hwcmp match: PX4FMU_V4
HMC5883_I2C on I2C bus 1 at 0x1e (bus: 100 KHz, max: 400 KHz)
HMC5883_SPI on SPI bus 1 at 5 (11000 KHz)
MPU6000 on SPI bus 1 at 6 (1000 KHz)
MPU9250 on SPI bus 1 at 4 (1000 KHz)
meas_airspeed: no MS4525 airspeed sensor connected
ets_airspeed: no ETS airspeed sensor connected
ets_airspeed: no ETS airspeed sensor connected
sensors: mag cal status changed 
sensors: mag cal status changed 
WARN  No RGB LED found at /dev/rgbled0
[i] FMU mode_pwm started 
ver hwcmp match: PX4FMU_V4
mavlink_if0: mode: 0, data rate: 800000 B/s on /dev/ttyS0 @ 921600B
mavlink_if0: hardware flow control not supported
mavlink_if0: offboard mission init: ERROR
mavlink_if1: mode: 0, data rate: 1200 B/s on /dev/ttyS2 @ 57600B
Processes: 22 total, 3 running, 19 sleeping
Processes: 22 total, 4 running, 18 sleeping
CPU usage: 35.14% tasks, 0.46% sched, 64.41% idle
Processes: 22 total, 3 running, 19 sleeping
Processes: 22 total, 3 running, 19 sleeping
Processes: 22 total, 4 running, 18 sleeping
CPU usage: 35.16% tasks, 0.46% sched, 64.38% idle
Uptime: 149.294s total, 99.329s idle

 PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE 
   0 Idle Task                   99328 64.378     0/    0   0 (  0)  READY 
   1 hpwork                       6819  4.652   784/ 1592 192 (192)  w:sem 
   2 lpwork                        551  0.381   356/ 1592  50 ( 50)  w:sig 
   3 init                         1462  0.000  1332/ 2496 100 (100)  w:sem 
 174 top                           184  3.279  1212/ 1696 100 (100)  RUN   
  85 dataman                         1  0.000   660/ 1192  90 ( 90)  w:sem 
 104 sensors                      5632  3.813  1740/ 1992 250 (250)  w:sem 
 106 gps                           219  0.152   660/ 1192 220 (220)  w:sem 
 108 commander                     710  0.381  3172/ 3592 140 (140)  w:sig 
 109 commander_low_prio              5  0.000   500/ 2872  50 ( 50)  w:sem 
 115 mavlink_if0                  6186  4.195  2372/ 2696 100 (100)  READY 
 116 mavlink_rcv_if0                11  0.000   812/ 2096 175 (175)  w:sem 
 119 mavlink_if1                  1278  0.839  2364/ 2696 100 (100)  w:sig 
 120 mavlink_rcv_if1                11  0.000   812/ 2096 175 (175)  w:sem 
 126 sdlog2                        459  0.305  2052/ 3296  70 ( 70)  w:sig 
 144 attitude_estimator_q         7089  4.805  1908/ 2392 250 (250)  w:sem 
 146 position_estimator_inav      3903  2.288  4748/ 5296 250 (250)  w:sem 
 153 mc_att_control               3363  2.288   828/ 1496 250 (250)  w:sem 
 155 mc_pos_control                902  0.610   964/ 1496 250 (250)  w:sem 
 159 navigator                     507  0.000   788/ 1496 105 (105)  w:sem 
 169 mavlink_if2                 10515  7.170  2372/ 2696 100 (100)  w:sig 
 171 mavlink_rcv_if2                20  0.000  1092/ 2096 175 (175)  w:sem 
WARN  navigator timed out

pkocmoud, 2016-01-28:
I found that if I reboot from NSH reboot command, the "WARN navigator timed out" appear within 5 minutes. Comfirmed twice. Otherwise on a cold power up, I did not get the message while waiting over 20 minutes.

I just noticed that the map shifted noticeably just before or while the message started.

pkocmoud, 2016-01-28:
Maybe this is just a GPS wandering issue. It has occured on 2 uBlox NEO m8 GPSs. I see a lot on movement tonight. I just do not recall ever seeing this message in the past.

LorenzMeier, 2016-01-28:
Ok. So fundamentally might be one of the things why we want to move to EKF2.

LorenzMeier, 2016-01-28:
@pkocmoud Could you get us a 200 Hz log from boot? The way to do this is:

```
sdlog2 stop
sdlog2 start -t -r 200 -e -b 32
```
Wait for the issue to happen
```
sdlog2 stop
```
Now its safe to power down the system and upload the log.

pkocmoud, 2016-01-28:
http://logs.uaventure.com/view/65pADCyf3jW5SuKSfMHU8D

pkocmoud, 2016-01-28:
Rerunning the test as ver all was not showing an accurate value.

LorenzMeier, 2016-01-28:
Ok, I understand it now. It's correct and harmless. Just tells you it lost lock.

closed 2016-01-28 by LorenzMeier