PX4/PX4-Autopilot #3586: Motors spinning after setting frame type

pkocmoud, 2016-01-25:
I built up a new PixRacer on my existing and known good Spedix 250. I wanted to get the wireless (non-usb) experience. I connected the battery and booted. I found the wifi network and joined. I set the frame type and starting parameters were the "3DR DIY Quad". I "apply and restart" using the button in QGC. 

Motors spun for approximately .5 seconds this first reboot. The issue has not occurred since. I was long and fast enough to make a mess of my desk. I have not tried to restart the process either.

LorenzMeier, 2016-01-25:
You should be using the QAV250 config. But independent of that the motor spin is of course an issue. I suspect its a result of the reboot with battery connected. Since the ESCs support OneShot even the slightest pulse during boot can trigger them. I need to scope the pin state.

LorenzMeier, 2016-01-25:
I found a 1400 us pulse on boot.

LorenzMeier, 2016-01-25:
And fixed it. Please re-test without props carefully and sorry about that - it has not been apparent luckily for anybody else.

LorenzMeier, 2016-01-25:
Needless to say, you need to update from master.

closed 2016-01-25 by LorenzMeier