PX4/PX4-Autopilot #3591: Pure fixed-wing startups fail with px4io mixer send error on master and stable

AndreasAntener, 2016-01-25:
@LorenzMeier

e.g. SYS_AUTOSTART 3031

```
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
[i] microSD mounted: /fs/microsd 
param: selected parameter default file /fs/mtd_params
[param] Loaded: /fs/mtd_params 
rgbled on I2C bus 2 at 0x55 (bus: 100 KHz, max: 100 KHz)
 1:   SYS_USE_IO: match
 3031: + SYS_AUTOSTART: match
px4io: CRCs match
dataman: Power on restart, data manager file '/fs/microsd/dataman' size is 103090 bytes
MS5611_SPI on SPI bus 1 at 3 (11000 KHz)
bst: no devices found
adc init done
ver hwcmp match: PX4FMU_V2
hmc5883: no device on bus 2
hmc5883: no device on bus 1
mpu6000: driver start failed
MPU6000 on SPI bus 1 at 4 (1000 KHz)
L3GD20 on SPI bus 1 at 1 (11000 KHz)
LSM303D on SPI bus 1 at 2 (11000 KHz)
meas_airspeed: no MS4525 airspeed sensor connected
ets_airspeed: no ETS airspeed sensor connected
ets_airspeed: no ETS airspeed sensor connected
px4io default PWM output device
mavlink_if0: mode: 0, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
mavlink_if0: offboard mission init: ERROR
ver hwcmp match: PX4FMU_V2
sdlog2: log buffer size: 12288 bytes
[i] FIXED WING 
commander_low_prio: settings saved.
mixer: fmu sent: " 10000 0 -10000 10000
"
px4io mixer send error -1
mixer: error loading mixers from /etc/mixers/phantom.main.mix
mixer: failed to load mixer
[i] Error loading mixer: /etc/mixers/phantom.main.mix 
pwm: failed get disarmed values

```

AndreasAntener, 2016-01-25:
Looks like our old friend of having to many mixer lines is back

LorenzMeier, 2016-01-25:
I don't quite understand why this would work for VTOL then though. Odd.

AndreasAntener, 2016-01-25:
The only thing I can see is that we have less mixer outputs in VTOL main mixers.

SimonWilks, 2016-01-26:
Looking further into this reveals the following:

Broken at:
```
commit 7005b18a66d1b75193403f6f658b57d2190efabb
Author: Lorenz Meier <lm@inf.ethz.ch>
Date:   Mon Dec 14 13:09:15 2015 +0000

    IO: Cleanup DSM input routine
```

Last working commit prior to the above commit:
```
commit 63da175b3fbd9f912b3332df01b628363a29e238
Author: Lorenz Meier <lm@inf.ethz.ch>
Date:   Wed Dec 23 12:47:37 2015 +0100

    Fix spurious mission warning
```

In between these two commits above I get compilation errors but it looks like these changes are IO related and are part of #3358 which was a DSM rewrite merged Dec. 24.

Note that this is only failing for larger mixer files which is most of the fixed-wings with gimbal mixers defined but also the Albatross which a large number of servos to drive.

Could it be related to IO memory?

SimonWilks, 2016-01-26:
Note that users will see the following message in QGC when this happens:

"You have connected to a Vehicle which does not have an SD Card inserted. Please insert and SD card and try again."

LorenzMeier, 2016-01-26:
@kd0aij Can you help looking into this?

LorenzMeier, 2016-01-26:
Ugh, IO went complete nuts:
```
nsh> px4io status
px4io: loaded
protocol 2147483648 hardware 2147483648 bootloader 2147483648 buffer 2147483648B crc 0x8000000080000000
2147483648 controls 2147483648 actuators 2147483648 R/C inputs 2147483648 analog inputs 2147483648 relays
2147483648 bytes free
status 0x0000 SAFETY_SAFE RC_FAIL FMU_FAIL MIXER_FAIL ARM_NO_SYNC INIT_FAIL
alarms 0x0000
vservo 2147483648 mV vservo scale 2147483648
vrssi 2147483648
actuators 0 0 0 0 0 0 0 0
servos 2147483648 2147483648 2147483648 2147483648 2147483648 2147483648 2147483648 2147483648
reversed outputs: [________] trims: r:   0.0000 p:   0.0000 y:   0.0000
0 raw R/C inputs
R/C flags: 0x0000
mapped R/C inputs 0x0000
ADC inputs
features 0x0000
arming 0x0000 FMU_DISARMED IO_ARM_DENIED
rates 0x80000000 default 2147483648 alt 2147483648
debuglevel 2147483648
controls 0: 0 0 0 0 0 0 0 0
controls 1: 0 0 0 0 0 0 0 0
controls 2: 0 0 0 0 0 0 0 0
controls 3: 0 0 0 0 0 0 0 0
input 0 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 1 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 2 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 3 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 4 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 5 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 6 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 7 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 8 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 9 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 10 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 11 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 12 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 13 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 14 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 15 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 16 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 17 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
failsafe 2147483648 2147483648 2147483648 2147483648 2147483648 2147483648 2147483648 2147483648
disarmed values 2147483648 2147483648 2147483648 2147483648 2147483648 2147483648 2147483648 2147483648
```

LorenzMeier, 2016-01-26:
Command to find wasted static allocations:
arm-none-eabi-nm --print-size --size-sort --radix dec -C /Users/lorenzmeier/src/Firmware/build_px4fmu-v2_default/src/modules/px4iofirmware/px4io-v2  | grep ' [DdBb] '

LorenzMeier, 2016-01-26:
Here is the fix: https://github.com/PX4/Firmware/pull/3603

AndreasAntener, 2016-01-27:
Fix went to master

closed 2016-01-27 by AndreasAntener