PX4/PX4-Autopilot #3600: Clamp manual yaw target when system is in yaw rate limit

LorenzMeier, 2016-01-26:
In this log the yaw tracking is really good:
http://logs.uaventure.com/view/xM8RPz26NeFtmcR8Cnm5RL#A_YawRate_PLOT

But the system response felt sluggish. This is most likely because a yaw setpoint was built up as the system was hitting the 60 deg/s yaw rate limit and so the quad continued to turn after the yaw set point.

priseborough, 2016-01-26:
I just flew https://github.com/PX4/Firmware/tree/yaw_man_limmit and it appears to have fixed the yaw overshoot behaviour i was seeing.
http://logs.uaventure.com/view/beciMrBZ8JVhDU9Z2vKDmQ

LorenzMeier, 2016-01-26:
Nice! Note that you might want to calibrate your ESCs or correct your system's CG: http://logs.uaventure.com/view/beciMrBZ8JVhDU9Z2vKDmQ#Actuators_PLOT

LorenzMeier, 2016-01-26:
Instructions: http://px4.io/docs/esc-calibration/

closed 2016-01-26 by julianoes

priseborough, 2016-01-26:
It has been calibrated, but I have a sick esc.

commit 5ea5ecf3 (2016-01-26): Limit manual yaw command properly. Fixes #3600