PX4/PX4-Autopilot #3613: frsky_telemetry not working on release v1.1.3

Diamg, 2016-01-27:
The app frsky_telemetry is not working in the new released version as it did on the release v1.1.2
Have you noticed that? There have been any significant change that could explain that? 
On previous releases just adding the the line:
```frsky_telemetry start -d /dev/ttyS2```
on the ```etc/extras.txt``` in the SD card was working.

Thanks

kd0aij, 2016-01-27:
Are you using a Pixhawk? Also, which series of FrSky receiver are you using
(X or D)?

kd0aij, 2016-01-27:
To answer your question: the frsky_telemetry daemon has been modified to add SmartPort telemetry, but it should work the same as before for D series telemetry. I've tested it on FMUv2 port ttyS2 using a D4R-II and Taranis Plus running OpenTX 2.1.

Diamg, 2016-01-28:
We are using a pixhawk with D4R-II on port ttyS2 but with a FlySky TH9X flashed by 3DR.

LorenzMeier, 2016-01-28:
Can you start the Frsky process from the console instead of from SD and share the output?
http://dev.px4.io/advanced-system-console.html
Please report back and share the output on the console. You have to make sure that you're not already starting it from the microSD.

robertgs90, 2016-01-28:
I had the same problem. I share a picture with my wiring. I am using the 3DR X8+. I tried the default connction to TELEM2 and using:
```frsky_telemetry start -d /dev/ttyS2```
to start telemetry.
When I do that from the shell and I check the status it says the telemetry is working but I recieve no signal in my RC.
I also tried to connect the telemetry to SERIAL 4/5 and ttyS6. 

[image]

If I try without -d I have an extra output line

```
nsh> frsky_telemetry start -d /dev/ttyS2
frsky_telemetry: opening uart
nsh> frsky_telemetry status
frsky_telemetry: running
nsh> frsky_telemetry stop
frsky_telemetry: .
frsky_telemetry: terminated.
nsh> 
nsh> frsky_telemetry start /dev/ttyS2
frsky_telemetry: opening uart
nsh> frsky_telemetry: SmartPort receiver not detected, sending FrSky D type telemetry

nsh> frsky_telemetry status
frsky_telemetry: running
nsh> frsky_telemetry stop
frsky_telemetry: .
frsky_telemetry: .
frsky_telemetry: .
frsky_telemetry: .
frsky_telemetry: terminated.
```
In both cases the telemetry is not working.  I don't know if when it says SmartPort receiver not detected it is wainting for some response

LorenzMeier, 2016-01-28:
There are two issues here:
 * You didn't wait in your first attempt for it to switch to D mode telemetry (which is your telemetry type)
 * You entered the wrong command on your 2nd attempt, which means it did not open /dev/ttyS2 but /dev/ttyS6

Can you re-try with the correct port and wait a little longer and provide the output of that?

LorenzMeier, 2016-01-28:
@kd0aij You might have to disable hardware flow control explicitly, it might be connected here. Hard to see.

robertgs90, 2016-01-28:
I tried again and wait more than a minute but there is no second output. 

If I try connecting to SERIAL 4/5, and executing:
```
nsh> frsky_telemetry start
frsky_telemetry: opening uart
nsh> 
```
I have no second output. It is not changing to D telemetry.

LorenzMeier, 2016-01-28:
Ok, that was a great result. So something is wrong with the switch.

@kd0aij The D switch heuristic probably needs to be reworked. Are you maybe not basing it wether you can decode something, but based on wether you get any data? It might be that you get the data you're writing yourself back. So you might need to either flush the buffer after every write, or try to decode the data and see if its a valid sensor hub message.

robertgs90, 2016-01-28:
Thanks a lot for your help.

kd0aij, 2016-01-28:
Yes, there must be traffic on the receive input of ttyS2. I just noticed that the instructions here: http://px4.io/docs/frsky-telemetry/ say to connect both RX and TX, but **a quick fix would be to disconnect the ttyS2 RX line** since it's not used by the original driver anyway, and that's the way I've been testing.
I'll look further into D type telemetry to see what's coming from the RX.

kd0aij, 2016-01-28:
The D series receiver is transmitting AD1, AD2 and "link quality" on its TX line. We could enhance the daemon to recognize this packet and perhaps also publish this data. Or we could just require that the D telemetry port TX line is not connected.

LorenzMeier, 2016-01-28:
I would much prefer a software solution if possible. The finicky TX line difference between the versions will let a lot of users stumble. But only you can judge if that's a quick fix or a major undertaking.

dogmaphobic, 2016-01-28:
Jumping into this thread from 30,000' high. From what I remember (well over a year ago), the FrSky RX sends *Requests* and the various sensors respond accordingly. If they have the requested sensor data type, they respond. I wrote a "sensor" using a Teensy back in 2014. I can look it up if needed.

kd0aij, 2016-01-28:
That's true for SmartPort. But not for "D" type telemetry.

kd0aij, 2016-01-28:
@LorenzMeier Should be a simple software fix to validate the D type incoming packet. Publishing RSSI to uORB should also be an easy enhancement, if there's an existing topic to publish to.

LorenzMeier, 2016-01-28:
Don't have that topic yet. Right now my main concern is to have people with D setups going again.

kd0aij, 2016-01-28:
:+1:  Firing up Eclipse

kd0aij, 2016-01-28:
software fix ready for testing: https://github.com/PX4/Firmware/pull/3624

LorenzMeier, 2016-01-28:
@kd0aij Nice!

robertgs90, 2016-01-29:
I tested the fixed version and D frsky telemetry is working well!
Thank you vey much!

LorenzMeier, 2016-01-29:
Cool! Merged to master, closing.

closed 2016-01-29 by LorenzMeier